knurling-rs/probe-run #8: What would it take to allow debugging?

matoushybl, 2020-08-17:
Hi,
thank you for the runner, it is something I've been waiting for a long time.

I ma just wondering what would need to be done to allow for debugging with standard IDEs, such as CLion, or VS code, in the past the runner was configured to arm-none-eabi-gdb.

I've seen that there are limitations in running RTT alongside with GDB in probe-rs, but I do not know enough to figure out the consequences for the runner.

Thanks!
I'll gladly contribute if the implementation will be something I can handle.

japaric, 2020-08-17:
Hello :wave: 

This particular runner, or at least in its current state, is meant to be used *instead* of GDB. The reason is that we have tied breakpoints to "exit"-ing the runner (and this also resets the device) so you would not be able to inspect the device with GDB once a breakpoint is reached.

> in the past the runner was configured to arm-none-eabi-gdb.

In VS code (+ Rust Analyzer) you can click the "Run" button (shortcut for `cargo-run`) instead of the "Debug" button to run programs with `probe-run`, after setting the runner to `probe-run` (in `.cargo/config`).

> there are limitations in running RTT alongside with GDB in probe-rs

Not familiar with the GDB implementation of probe-rs but in principle I don't see GDB and RTT being incompatible (GDB degrading RTT's performance is expected though).

`defmt` logs (showcased in [this blog post]) are not tied to RTT and can be used with other transports (e.g. serial interface, USB or ITM) so worst case scenario (GDB & RTT not working together at all -- unlikely I think) you'll be able to get logs out of your device using a different transport.

[this blog post]: https://ferrous-systems.com/blog/21st-century-embedded-tooling/

---

That being said, it should be possible to have another runner (or add a second mode to `probe-run`) that supports the 'exit' feature and also the standard GDB interface.

What kind of workflow did you have in mind? `defmt` logs and being able to exit with 0 and non-zero exit code, but with breakpoints and backtraces from GDB? (that one should be possible)

matoushybl, 2020-08-19:
Thanks for the clarification on what exactly is the purpose of the runner functionality wise.

What I meant by debugging was more of a seamless debugging integration - like clicking on the bug icon and the IDE correctly launching the arm-*-gdb and loading the program, but I see now that the previous solution with gdb scripts works just as well.

Thank you for the runner and the effort of making rust embedded the best!

closed 2020-08-19 by matoushybl