knurling-rs/probe-run #15: Unable to flash STM32H7 MCU

mattico, 2020-08-18:
```
Running `probe-run --chip STM32H743BITx --defmt target\thumbv7em-none-eabihf\release\mainboardfw-rs`
flashing program ..
Error: Error while flashing

Caused by:
    No flash memory contains the entire requested memory range 0x8000000..0x8005904.
```

This is caused by probe-rs having an incorrect memory map for the flash, which was fixed in https://github.com/probe-rs/probe-rs/pull/314.

japaric, 2020-08-19:
Looking at the dates it seems that PR hasn't made it into a crates.io release. If it gets released as part of the 0.8.x series then reinstalling `probe-run` should fix this issue.

tstellanova, 2020-09-15:
It appears the upstream fix was released as part of probe-rs version 0.9.0, so this issue still appears with probe-run 0.1.3

diondokter, 2020-10-15:
Yup, same issue here. Would love to see it updated

noppej, 2020-11-04:
> It appears the upstream fix was released as part of probe-rs version 0.9.0, so this issue still appears with probe-run 0.1.3

@tstellanova ... is there any way for me to update my environment to to run the fixed probe-rs with probe-run, or do I need to wait for a new release?

tstellanova, 2020-11-04:
I have not tried this, but I suppose that
modifying probe-run’s cargo.toml
to depend on the version of probe-rs desired
and then build & run should work?

jonas-schievink, 2020-11-04:
`cargo install --git https://github.com/knurling-rs/probe-run.git --branch update` should work for now

noppej, 2020-11-04:
> `cargo install --git https://github.com/knurling-rs/probe-run.git --branch update` should work for now

@jonas-schievink ... thanks for your speedy reply, but this fails for me with the same error I got when trying @tstellanova 's suggestion. i.e.

```
<snip>  
Compiling probe-rs-rtt v0.4.0  

<snip>let me know if you want full ld command line<snip>  

ld: library not found for -lftdi1
          clang: error: linker command failed with exit code 1
```

PS. I am running on Mac OSX (latest version)

jonas-schievink, 2020-11-04:
Yes, that's why it isn't merged yet :) You'll have to install libftdi by hand.

noppej, 2020-11-04:
Excellent. A simple `brew install ftdi` solved the `ld` error.

I can also confirm that this combo fixes the "Unable to flash STM32H7 MCU" issue for my board. I can now flash, run and backtrace.

Thanks for all your help.

japaric, 2020-11-23:
`probe-run` 0.1.6, which depends on probe-rs 0.10.0, has just been published. This should be fixed in that release so I'm going to close this PR. If you still run into problems feel to open a new issue.

closed 2020-11-23 by japaric