knurling-rs/probe-run #22: What would it take to support Teensy 3 & 4?

Walther, 2020-08-19:
What would it take to support the Teensy series of chips - especially the newer 3.x and 4.x families based on Cortex M families?

- Teensy 4.0 and 4.1: "an ARM Cortex-M7 processor at 600 MHz, with a NXP iMXRT1062 chip, the fastest microcontroller available today"
- Teensy 3.6: "a 32 bit 180 MHz ARM Cortex-M4 processor with floating point unit"
- Teensy 3.5: "a 32 bit 120 MHz ARM Cortex-M4 processor with floating point unit"
- (Teensy 3.1 and 3.2: "a 32 bit ARM processor")
- (Teensy 2.0: AVR)

Some possibly-relevant work e.g.
- [teensy 4](https://github.com/mciantyre/teensy4-rs)
- teensy 3.5, 3.6 ?
- [teensy 3.1, 3.2](https://github.com/jamesmunns/teensy3-rs) "This crate is meant to be consumed by a binary for a PJRC Teensy 3.1 or 3.2. Support for Teensy 3.0, 3.5, and 3.6 boards is planned, but not implemented."

The newer Teensy chips provide a fun amount of compute power not commonly seen in other boards, I'd love to see better Rust support for these boards ❤️

japaric, 2020-08-20:
I believe the older models of the Teensy don't have an on-board debugger but the newer ones do? The ones that do have one should already work with `probe-run`.

In general, boards that have no on-board debugger have some bootloader mode. It is certainly possible to write a Cargo runner that loads programs using the bootloader and then sets up a `defmt` transport but that usually requires quite a bit of device, or even board, specific code. What would it take to support those boards? I'm afraid that I cannot answer as I'm not familiar with this particular hardware (board or chip).

jamesmunns, 2020-11-17:
I don't believe any of these boards have debugger access, and all have an external bootloader device.

We would instead need to implement defmt-over-usb support for these, or someone could implement a specific defmt sink for this.

cstrahan, 2022-09-25:
FWIW, there are some hardware mods that can make JTAG/SWD work on Teensy.

See: https://forum.pjrc.com/threads/69748-Simple-WORKING-JTAG-UART-for-Teensy-4-1

My understanding from perusing the forums a while back is that the ARM chips on Teensy have a fuse burnt during production that selects for JTAG interface (as, IIUC, the MKLO2 on the Teensy uses JTAG to control the processor during power on, and probably in some other cases I'm unaware of), and _not_ SWD -- but it would seem that you can do one of two things if you want/need SWD:

* Use JTAG to poke the register that selects between JTAG and SWD, and/or
* Very early in your code, configure that register as desired -- now you'll have SWD shortly after power-on (at least until you flash another program _without_ that bit of configuration code).

I have a couple Teensies and I hope to make such a modification to them so I can develop and test/debug some Rust code on them, in which case maybe I'll report back when I do :slightly_smiling_face:.

Urhengulas, 2022-10-03:
Please share if you suceed, but also if you find any blockers @cstrahan!