knurling-rs/probe-run #26: Runner hangs at resetting device when using async-embedded.

matoushybl, 2020-08-19:
Hi,
I know that async is extremely experimental, however when exprimenting with it, I stumbled upon the problem that when I was trying to run the code wit probe-run nothing was printed using RTT, but when I ran it using cargo embed, RTT work normally.
The code with respective outputs is here:
```
#![no_std]
#![no_main]

use panic_halt as _; // you can put a breakpoint on `rust_begin_unwind` to catch panics

use cortex_m::asm::bkpt;
use cortex_m_rt::entry;
use rtt_target::{rprintln,rtt_init_print};
use async_embedded::task;

#[entry]
fn main() -> ! {
    rtt_init_print!();
    rprintln!("Hello, world!");

    task::spawn(async move {```
        loop {
            rprintln!("a");
            task::r#yield().await;
        }
    });

    task::block_on(async {
        loop {
            rprintln!("b");
            task::r#yield().await;
        }
    })
}
```
probe-run (when stopped):
```
flashing program ..
DONE
resetting device
stack backtrace:
   0: 0x08000410 - core::future::get_context
   1: 0x08000a8c - async_embedded::task::yield::{{closure}}
   2: 0x0800092c - <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
   3: 0x080015a2 - runner_test::__cortex_m_rt_main::{{closure}}
   4: 0x080008c8 - <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
   5: 0x08000c60 - async_embedded::executor::Node<dyn core::future::future::Future+Output = ()>::new::{{closure}}
   6: 0x08000864 - <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
   7: 0x08000f84 - async_embedded::executor::Executor::block_on
   8: 0x08000b48 - async_embedded::task::block_on
   9: 0x08001254 - runner_test::__cortex_m_rt_main
  10: 0x080010a8 - main
  11: 0x08003c76 - ResetTrampoline
  12: 0x08003c6c - Reset
Error: RTT control block not found in target memory. Make sure RTT is initialized on the target.

Process finished with exit code 1
```
Cargo embed: correctly showing alternating 'a' and 'b'.

japaric, 2020-08-20:
> Runner hangs at resetting device

Our error handling may be a bit off. This scenario corresponds to the case where RTT logging is not available (or detected as not available) and you should get this message: "No log messages to print, waited for device to halt" before `probe-run` blocks.

`probe-run` not correctly detecting the availability of RTT logs may have already been fixed in #25. I probably can make a new release with that included soon. In the meantime I would suggest running *optimized* programs (`cargo run --release`); that should in your case I think.

If running an optimized program still doesn't work could you point to the exact source code you are running? Either that or you could run `rust-nm -C target/thumbv7em-none-eabi/debug/app-name` (and `release` folder too) and let me know if `_SEGGER_RTT` appears in the output.

matoushybl, 2020-08-21:
Running the code in release mode worked, both the resulting binaries contain the `_SEGGER_RTT` section.

Thanks, will try with the new version when available.

japaric, 2020-11-16:
@matoushybl is this still an issue with the crates.io version of defmt and/or probe-run v0.1.4?

matoushybl, 2020-11-16:
I'll check it asap.

matoushybl, 2020-11-16:
I believe that this issue can be closed for now. I ran the async-on-embedded only as an experiment and I don't have the source code now. If I ever try it again and it happens, I'll definitely report it here.

closed 2020-11-16 by matoushybl