knurling-rs/probe-run #36: Error: `_defmt_*` symbol not found

mattico, 2020-08-26:
```
> probe-run --chip STM32H743BITx --defmt .\target\thumbv7em-none-eabihf\debug\mainboardfw-rs
Error: `_defmt_*` symbol not found
````

Probably related: https://github.com/knurling-rs/defmt/issues/132

Using defmt HEAD.

I'm building probe-run HEAD with probe-rs HEAD to work around #15.
```toml
[patch.crates-io]
probe-rs = { git = "https://github.com/probe-rs/probe-rs", rev = "0f3a2856365304dd6c04fc8c28dc82f8ac06e8bb" }
probe-rs-rtt = { git = "https://github.com/probe-rs/probe-rs-rtt" }
```

My firmware is linked with GCC like so:

```toml
[build]
target = "thumbv7em-none-eabihf"
rustflags = [
  "-C", "inline-threshold=225",
  "-C", "linker=arm-none-eabi-gcc",
  "-C", "link-arg=-Wl,-Tlink.x",
  "-C", "link-arg=-Wl,-Tdefmt.x",
  "-C", "link-arg=-nostartfiles",
  "-C", "link-arg=-specs=nano.specs",
  "-C", "link-arg=-lc",
  "-C", "link-arg=-mcpu=cortex-m7",
  "-C", "link-arg=-mfpu=fpv5-d16",
  "-C", "link-arg=-mfloat-abi=hard",
  "-C", "link-arg=-mthumb",
]
```

The symbols do seem to be present:
```
> arm-none-eabi-readelf.exe --sections --symbols .\target\thumbv7em-none-eabihf\debug\mainboardfw-rs | grep -i defmt
  [13] .defmt            PROGBITS        20001f1c 020218 00001f 00      0   0  1
  7363: 08006941   302 FUNC    LOCAL  DEFAULT    2 _ZN50_$LT$defmt_rtt..Logg
  7370: 08000993    82 FUNC    LOCAL  DEFAULT    2 _ZN5defmt9Formatter3fmt17
  7371: 08006903    62 FUNC    LOCAL  DEFAULT    2 _ZN5defmt9Formatter4istr1
  7372: 08006857   172 FUNC    LOCAL  DEFAULT    2 _ZN5defmt9Formatter5leb64
  7384: 2000021d     1 OBJECT  LOCAL  DEFAULT    5 _ZN9defmt_rtt17INTERRUPTS
  7385: 2000021c     1 OBJECT  LOCAL  DEFAULT    5 _ZN9defmt_rtt5TAKEN17h13d
  7386: 20001b1c  1024 OBJECT  LOCAL  DEFAULT   12 _ZN9defmt_rtt6handle6BUFF
  8134: 20001f28     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_info_start
  8135: 20001f1c     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_error_start
  8153: 20001f31     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_trace_start
  8163: 20001f28     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_warn_end
  8168: 20001f1d     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_version_ = 82dd53f
  8221: 080010c3    36 FUNC    GLOBAL DEFAULT    2 _defmt_timestamp
  8247: 08006a71   102 FUNC    GLOBAL DEFAULT    2 _defmt_acquire
  8269: 20001f27     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_warn_start
  8293: 20001f33     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_trace_end
  8354: 08006ad7    32 FUNC    GLOBAL DEFAULT    2 _defmt_release
  8363: 20001f2b     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_debug_start
  8412: 20001f27     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_error_end
  8428: 20001f31     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_debug_end
  8515: 20001f2b     0 NOTYPE  GLOBAL DEFAULT   13 _defmt_info_end
```

Going to try searching for the source of the error message.

mattico, 2020-08-26:
The error comes from elf2table. BTW `cargo vendor` is very useful for searching dependencies.

jonas-schievink, 2020-08-26:
> linked with GCC

There are no quotes around `_defmt_version_ = 82dd53f`, which we currently expect to be present (and they are, with LLD). Looks like that's a bug in LLD then. The fix/workaround is to look for the symbol both with and without quotes.

mattico, 2020-08-26:
Ah, okay. I'll put together a patch for that.

jonas-schievink, 2020-08-26:
Could you try with https://github.com/knurling-rs/defmt/pull/138?

closed 2020-08-27 by jonas-schievink