knurling-rs/probe-run #70: Support multiple applications

jamesmunns, 2020-09-22:
We've had folks request the ability to use defmt in a bootloader AND an application, which have unrelated defmt message libraries. Users also seem to want to reuse the same defmt RTT control block and memory space.

MathiasKoch, 2020-09-22:
I for one would love to see this be an option!

jamesmunns, 2020-09-22:
Relevant chat logs: https://freenode.logbot.info/rust-embedded/20200922#c5187053

therealprof, 2020-09-22:
May I suggest you amend the title to make it clear that this concerns multiple applications on a device (or in case of a bootloader probably even staged applications) and not multiple applications in a project (which lured me here because it works just fine)?

jamesmunns, 2020-09-22:
Thanks @therealprof, does that look more clear now?