knurling-rs/probe-run #72: Detect when the `rt` feature was not enabled

jonas-schievink, 2020-09-22:
I'm not sure if this can be detected, but if it can, we should totally try to do it, since everyone forgets to enable the `"rt"` feature from time to time and the symptoms are just "stuff's broken yo", which isn't very easy to debug.

japaric, 2020-11-16:
isn't this already caught at link time by an assertion in cortex-m-rt linker script?

matoushybl, 2021-01-23:
I forgot to enable the feature yesterday and the symptoms were just that interrupts handlers were not being called and the CPU getting stuck. I would love to help with a PR with a fix but I don’t have enough background to figure out the detection.

japaric, 2021-02-04:
> Detect when the `rt` feature was not enabled

would the goal be printing a warning when it's not enabled? the 'rt' is not required unless you use interrupts, right?
earlier this week, I ran into the issue of *not* enabling the rt feature in the stm32l4xx-hal and using hardware tasks in rtic: I didn't get any sort of linker error but none of the interrupts would fire (I think all of rtic's interrupt handlers got GC-ed by the linker). unsure if that's something that should be reported by rtic or cortex-m-rt ...

Urhengulas, 2021-05-28:
What `rt` feature of which crate are you referring to? 😇

jonas-schievink, 2021-05-28:
PACs generally have a feature called "rt", for example here: https://github.com/nrf-rs/nrf52840-pac/blob/master/Cargo.toml

Urhengulas, 2021-05-29:
> PACs generally have a feature called "rt", for example here: https://github.com/nrf-rs/nrf52840-pac/blob/master/Cargo.toml

So the idea is to detect the feature for any pac crate? That sounds more complicated that I though it is 🤔

japaric, 2021-07-01:
if you depend on a PAC but do *not* enable its `rt` feature
- there's no `#[cortex_m_rt::interrupt]` attribute so you cannot register an interrupt handler that will be discarded by the linker -> Good
- if you enable `cortex-m-rt`'s `device` feature (the `rt` feature of PACs enables this feature), you still cannot use the `#[cortex_m_rt::interrupt]` attribute; you cannot use `#[$PAC::interrupt]` either

I think that (`cortex-m-rt` API) works as intended.

I think the issue is that RTIC does *not* internally use the `#[cortex_m_rt::interrupt]` interface (I don't remember the reason for that), instead it directly defines symbols using `#[no_mangle]`. If the PAC's `rt` feature is not enabled then all the interrupt handlers generated by RTIC will be discarded by the linker leading to interrupts misbehaving.

So *I* think that the issue should be fixed in RTIC.

I don't think we can detect the issue in `probe-run` because we have the ELF *after* linking.

For starters, *not* enabling the PAC's `rt` feature is not a problem per se. If you are *not* using interrupt handlers then things will work. `probe-run` should not reject these binaries.

The issue is declaring interrupt handlers like RTIC does (not using the `#[cortex_m_rt::interrupt]` macro). Those handlers will be discarded by the linker if the "rt" crate is not enabled. When the linked ELF is passed to `probe-run` we can tell it has no interrupt handlers but cannot tell if the user defined them to begin with (because they are discarded by the linker).

> and the symptoms are just "stuff's broken yo", which isn't very easy to debug.

is stuff broken only with RTIC or are there other things that break when you forget to enable the "rt" feature? @jonas-schievink

japaric, 2021-07-01:
RTIC 0.6 (unreleased) does report the issue:
``` console
error[E0433]: failed to resolve: could not find `interrupt` in `you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml`
  --> src/bin/hello.rs:16:1
   |
16 | #[rtic::app(device = nrf52840_pac)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ could not find `interrupt` in `you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml`
   |
   = note: this error originates in an attribute macro (in Nightly builds, run with -Z macro-backtrace for more info)
```
I don't know what the timeline for that is but maybe that Cargo feature detection feature can be backported to 0.5.x

jonas-schievink, 2021-07-01:
> is stuff broken only with RTIC or are there other things that break when you forget to enable the "rt" feature? @jonas-schievink

That is a good question, I don't remember how I ran into this when I filed this issue

japaric, 2021-07-06:
> I don't know what the timeline for that is but maybe that Cargo feature detection feature can be backported to 0.5.x

this "you forgot to enable the 'rt' feature in the PAC" feature has been backported and released on version 0.5.7.

@jonas-schievink given the above maybe we can close this issue until we ran into something non-rtic related?

Urhengulas, 2023-02-11:
Closing as it seems fixed in RTIC and no other case where this is an issue was reported. If you hit this issue please open a new issue and mention this one here.

closed 2023-02-11 by Urhengulas