knurling-rs/probe-run #74: display filter via env variable

japaric, 2020-10-07:
use case: `defmt` supports emission filters on the target side (via Cargo features like `defmt-info`) but these have crate granularity. To filter that further at module level granularity we can use an env variable like [env_logger's `RUST_LOG`](https://docs.rs/env_logger/0.7.1/env_logger/#enabling-logging).

user interface: We can reuse `RUST_LOG`'s syntax: `krate::module=level,krate2::module2=level2` for familiarity. 

implementation: log messages include the module in their call site info. `probe-run` can read the env var, parse it, filter out incoming log messages and print to the console only those that pass the filter.

unresolved questions:
- env var name? `DEFMT_DISPLAY_FILTER`? `PROBE_RUN_FILTER`?

``` console
$ DEFMT_DISPLAY_FILTER=krate::module=info cargo run --bin my-app
0.000000 INFO  message1
└─ krate::module::function1 @ src/module/foo.rs:8
0.000001 INFO  message2
└─ krate::module::function2 @ src/module/bar.rs:9
(..)
```

Urhengulas, 2021-06-28:
> * env var name? `DEFMT_DISPLAY_FILTER`? `PROBE_RUN_FILTER`?

What about just `DEFMT_FILTER`? I don't see any other filters for defmt or probe-run. Then we would have `DEFMT_LOG` for the target-side log-level and `DEFMT_FILTER` for the host-side log-level.

japaric, 2021-07-15:
> We can reuse RUST_LOG's syntax: krate::module=level,krate2::module2=level2 for familiarity.

[`env_logger` also supports regex filters](https://docs.rs/env_logger/0.9.0/env_logger/#filtering-results). That's something we could implement in this feature, on the host (instead of on the target, see knurling-rs/defmt#519)

Urhengulas, 2023-02-11:
@japaric Since `DEFMT_LOG` is able of module level filtering, do you think we actually still want this feature? I am unsure.