knurling-rs/probe-run #75: Text not printed with rprint without newline

jiayihu, 2020-10-13:
I have a strange bug where `rprint!(".")` doesn't print anything until I print a newline `\n` later. This doesn't happen with cargo-embed so I created the issue here

japaric, 2020-10-19:
Thanks for the report! Applications that print to the terminal will usually only flush after a newline. When `--defmt` is not used we should `flush` after each `write` to the `stdout`. So a `flush` after https://github.com/knurling-rs/probe-run/blob/df90d57c8e1e49b6a27df27a0769619ba032084e/src/main.rs#L430 should fix this issue.

japaric, 2020-10-19:
@jiayihu could you try the latest (git) version of `probe-run` and let us know if this was fixed?

Re-opening until we heard back from the author

jiayihu, 2020-10-19:
I'm currently stuck with other issues in the toolchain, I'll let you know ASAP!

jiayihu, 2020-10-20:
Yeah works fine now and I think I'll stick with the git version, logs are prettier now 😄

closed 2020-10-20 by jiayihu

japaric, 2020-10-20:
Thanks for confirming!

PR #77 (2020-10-19): flush after each stdout write when defmt is not enabled
terminal apps flush on newline by default (changing that is not straightforward
iirc)
so instead we flush after each write call so that data not containing a newline
is printed immediately
should fix #75
commit 2f450303 (2020-10-19): flush after each stdout write when defmt is not enabled

terminal apps flush on newline by default (changing that is not straightforward
iirc)
so instead we flush after each write call so that data not containing a newline
is printed immediately
should fix #75