knurling-rs/probe-run #76: UDF and HardFault do not trigger an exception stopping probe-run on thumbv6m-none-eabi.

matoushybl, 2020-10-18:
When a HardFault is triggered by the UDF instruction, the processor moves to the infinite loop defined in cortex-m crate and the application does not terminate.
Is this the correct behavior?

japaric, 2020-10-19:
> Is this the correct behavior?

No, the intended behavior is that reaching the `HardFault` handler triggers a (hardware) breakpoint. `probe-run` sees that and exits the application (regardless of what the `HardFault` handler does)

There have been bugs around this in the past so this could be another one. One bug we fixed upstream in probe-rs/probe-rs#368 but the fix hasn't made it to a crates.io release.

Please give me the following information so I can narrow down the issue:
- What microcontroller is this? Share the argument you pass to the `--chip` flag
- What probe are you using? e.g. STLINK or JLINK or a CMSIS-DAP probe. You can report the output of `probe-rs-cli list` for this bullet. Install [`probe-rs-cli`](https://crates.io/crates/probe-rs-cli) and run the command with the microcontroller connected to your PC.
- What's the address of the `HardFault` symbol? If you run your program with `cargo run --example foo`, you can run `cargo nm --example foo -- -- -CSn` and report all lines that contain `HardFault`. `cargo-nm` is part of the [`cargo-binutils`](https://crates.io/crates/cargo-binutils) crate.
- If you can share the binary you are trying to run that would help a lot. It should be in `target` directory and its path is printed when you run `cargo run --bin hello`: `probe-run --chip LPC845M301JBD48 --defmt target/thumbv6m-none-eabi/debug/hello` (the last argument).

matoushybl, 2020-10-20:
Hi,
here are the requested information:
* stm32f031f6px, but also happens on a F091
* Chinese STLink v2 clone `STLink V2 (VID: 0483, PID: 3748, Serial: 361213002C135737334D4E00, STLink)`
* ```
08001991 T HardFault
0800197c T HardFaultTrampoline
08001990 T HardFault_
```
* https://drive.google.com/file/d/1yDXrQfiLpZibYboTXQV_80x4tOKNxizr/view?usp=sharing

Thanks for your help!

japaric, 2020-10-21:
@matoushybl are you using the crates.io release (v0.1.3) of `probe-run`? or using a git install? I behavior I described in https://github.com/knurling-rs/probe-run/issues/76#issuecomment-712098805 is available in the git version but we haven't made a crates.io release that includes that change.

I just checked v0.1.3 (crates.io version) of probe-run and yes `probe-run` gets stuff and doesn't not terminate when the microcontrollers hits UDF or HardFault.

matoushybl, 2020-10-21:
I’ll check it later today, but I believe I am using the git one because of defmt.

matoushybl, 2020-10-24:
I have just checked that this happens with a freshly installed git version of probe run `probe-run 0.1.3 (92a60cf 2020-10-23)`.

matoushybl, 2020-11-16:
The current version triggers the exception output but doesn't terminate probe-run, it gets stuck with the output: 
``` 
 (HOST) INFO  flashing program
  (HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
0.000000 ERROR panicked at 'attempt to divide by zero', src/driver.rs:253:21
└─ panic_probe::print_defmt::print @ /home/matous/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.1.0/src/lib.rs:140
```
If this is the correct implementation, then this issue can be closed.

jonas-schievink, 2020-11-16:
Just to make sure, you are using `panic-probe`, and no other panic handlers, right?

jonas-schievink, 2020-11-16:
Oh it's right in the path, duh.

Yeah this does seem like a bug then.

matoushybl, 2020-11-16:
No other panic handlers that I am aware of, the project originates in the app-template template.

japaric, 2020-11-16:
One thing that seemed suspicious in the provided ELF binary is that the `objdump` output shows that the addresses have the thumb bit set (`HardFault` routine) but I didn't see that in a local build (but for the M4F target). I'll check this, this week; I now have access to some M0 devices.

daniel-svard, 2020-11-16:
I have the same problem on an STM32G071 microcontroller. I found that the problem is that probe-rs does not set the memory address of the breakpoint for M0 devices correctly. It masks out 8 instead of 5 bits, causing the breakpoint to be placed at the wrong address when setting one at high address (the flash on STM32 devices start at 0x0800_0000).

I have a fix for it in my fork of [probe-rs](https://github.com/daniel-svard/probe-rs/commit/659f42bf8fabbfbbaafa4620f2e460dae79681ef), but I have not yet filed a pull-request upstream.

However, there is another issue ([probe-rs/probe-rs#425](https://github.com/probe-rs/probe-rs/issues/425)) in probe-rs/master (which I used as base for my fix) that causes probe-rs to panic for me. I suppose you can try to cherry-pick my fix onto a previous revision of probe-rs. I used the merge commit of @japaric pull-request [probe-rs/probe-rs#368](https://github.com/probe-rs/probe-rs/pull/368) to test it before I rebased it onto master.

japaric, 2020-11-17:
@daniel-svard thanks for investigating!
@matoushybl you may want to test out Daniel's fork to see if that fixes the issue for you (you'd need to rebuild probe-run against the fork)

tentatively re-labeling this as 'status: blocked' (on upstream). I hope the fixes make it into the next probe-rs release!

daniel-svard, 2020-11-17:
The fix is now merged into upstream ([ccb371e](https://github.com/probe-rs/probe-rs/commit/ccb371e92cb9ae815ce0bb612dbc9817af49979a)) so it will probably make it into the next probe-rs release.

Sh3Rm4n, 2020-12-20:
Isn't this resolved with #108?

matoushybl, 2020-12-21:
I can confirm that this is now working on STM32F091 when attempting to divide by zero.

closed 2020-12-21 by matoushybl