knurling-rs/probe-run #80: Build failure with defmt on current main head

daschl, 2020-10-22:
**Describe the bug**
Since last night I can't compile it because of some issues with defmt code it calls.

This is on OSX, probe-run hash `4ee681e`

**To Reproduce**
``` rust
cargo install probe-run --git https://github.com/knurling-rs/probe-run.git --branch main -f --features defmt
```

Output:

```
 ... all fine until here ...
  Compiling probe-rs-t2rust v0.6.0
   Compiling defmt-elf2table v0.1.0 (https://github.com/knurling-rs/defmt?branch=main#91f33fc4)
   Compiling probe-rs v0.8.0
   Compiling probe-rs-rtt v0.3.0
   Compiling probe-run v0.1.3 (/Users/daschl/.cargo/git/checkouts/probe-run-31a04fec2ca67672/4ee681e)
error[E0425]: cannot find function `parse` in crate `defmt_elf2table`
   --> src/main.rs:140:38
    |
140 |         let table = defmt_elf2table::parse(&bytes)?;
    |                                      ^^^^^ not found in `defmt_elf2table`

error[E0425]: cannot find function `get_locations` in crate `defmt_elf2table`
   --> src/main.rs:150:41
    |
150 |             let locs = defmt_elf2table::get_locations(&bytes, table)?;
    |                                         ^^^^^^^^^^^^^ not found in `defmt_elf2table`

error[E0603]: enum `Level` is private
  --> src/logger.rs:32:24
   |
32 |         defmt_decoder::Level::Trace => Level::Trace,
   |                        ^^^^^ private enum
   |
note: the enum `Level` is defined here
  --> /Users/daschl/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/91f33fc/decoder/src/lib.rs:21:30
   |
21 | use defmt_parser::{Fragment, Level, Type};
   |                              ^^^^^

error[E0603]: enum `Level` is private
  --> src/logger.rs:33:24
   |
33 |         defmt_decoder::Level::Debug => Level::Debug,
   |                        ^^^^^ private enum
   |
note: the enum `Level` is defined here
  --> /Users/daschl/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/91f33fc/decoder/src/lib.rs:21:30
   |
21 | use defmt_parser::{Fragment, Level, Type};
   |                              ^^^^^

error[E0603]: enum `Level` is private
  --> src/logger.rs:34:24
   |
34 |         defmt_decoder::Level::Info => Level::Info,
   |                        ^^^^^ private enum
   |
note: the enum `Level` is defined here
  --> /Users/daschl/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/91f33fc/decoder/src/lib.rs:21:30
   |
21 | use defmt_parser::{Fragment, Level, Type};
   |                              ^^^^^

error[E0603]: enum `Level` is private
  --> src/logger.rs:35:24
   |
35 |         defmt_decoder::Level::Warn => Level::Warn,
   |                        ^^^^^ private enum
   |
note: the enum `Level` is defined here
  --> /Users/daschl/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/91f33fc/decoder/src/lib.rs:21:30
   |
21 | use defmt_parser::{Fragment, Level, Type};
   |                              ^^^^^

error[E0603]: enum `Level` is private
  --> src/logger.rs:36:24
   |
36 |         defmt_decoder::Level::Error => Level::Error,
   |                        ^^^^^ private enum
   |
note: the enum `Level` is defined here
  --> /Users/daschl/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/91f33fc/decoder/src/lib.rs:21:30
   |
21 | use defmt_parser::{Fragment, Level, Type};
   |                              ^^^^^

error: aborting due to 7 previous errors

Some errors have detailed explanations: E0425, E0603.
For more information about an error, try `rustc --explain E0425`.
error: failed to compile `probe-run v0.1.3 (https://github.com/knurling-rs/probe-run.git?branch=main#4ee681e6)`, intermediate artifacts can be found at `/var/folders/2t/p_gqr4sn1qb75zjtt673kx000000gr/T/cargo-installLdrOuP`

Caused by:
  could not compile `probe-run`.

To learn more, run the command again with --verbose.
```

fnafnio, 2020-10-22:
I think this issue arises as a side effect of https://github.com/knurling-rs/defmt/commit/a92fd1948ba88a2510da741fe3464ec01a394829

daschl, 2020-10-22:
yeah looks like it - btw great to see more people in austria hack on this stuff @fnafnio

jonas-schievink, 2020-10-22:
Hmm, why doesn't the lockfile work? It should pin defmt to a version that works.

Lotterleben, 2020-10-22:
Argh, I think that's because due to our `unstable` feature hack– I've opened https://github.com/knurling-rs/probe-run/pull/81 to hopefully fix this and detect hickups like these earlier in the future

closed 2020-10-22 by jonas-schievink

daschl, 2020-10-22:
are there any further changes required? probe-rs compiles fin, but when I run it now I get

```
   Compiling nrf52840dk-sample v0.1.0 (/Users/michaelnitschinger/tmp/nrf52840dk-sample)
    Finished dev [unoptimized + debuginfo] target(s) in 0.38s
     Running `probe-run --chip nRF52840_xxAA --defmt target/thumbv7em-none-eabihf/debug/main`
Error: `_defmt_*` symbol not found
```

jonas-schievink, 2020-10-22:
You need to stay on the old defmt version (https://github.com/knurling-rs/defmt/commit/be7f4a7cbe) for now, probe-run has not yet been updated to the latest defmt version.

daschl, 2020-10-22:
Yes that works, thanks!