knurling-rs/probe-run #86: Allow GDB connection

nosense-cz, 2020-10-26:
I was wondering whether it will be possible to include gdb-server stub from probe-rs to probe-run. The idea is to allow GDB connection and use regular debugging while maintain display of defmt log messages. 

When the probe-run is running it is not possible to access the st-link probe from the gdb-server nor OpenOCD. So to allow this to work a GDB session has to be somehow integrated into the probe-run.

jonas-schievink, 2020-10-26:
> The idea is to allow GDB connection and use regular debugging while maintain display of defmt log messages.

I think probe-rs currently does not support multiplexing the connection like that, but this would indeed be nice to have!

nosense-cz, 2020-11-02:
I started thinking about alternative defmt host that will use openocd as it's backend. I will try to hack together a proof of concept in next few days. So far I proved I can control target using telnet interface when a gdb session is active.

japaric, 2021-01-05:
update: tried the git version of `cargo-embed` today and it supports RTT and GDB server running concurrently so presumably this should be supported in the git version of probe-rs. cargo-embed 0.10.1 does not support the concurrent connection.

rursprung, 2022-10-30:
> update: tried the git version of `cargo-embed` today and it supports RTT and GDB server running concurrently so presumably this should be supported in the git version of probe-rs. cargo-embed 0.10.1 does not support the concurrent connection.

this feature had been released in june 2021 with `cargo-embed` 0.11.0, see [their blog post](https://probe.rs/blog/release-0-11-0/) and the underlying support must accordingly be in a released version of `probe-rs`.
it'd be great if it'd be possible to use `probe-run` with GDB and defmt-rtt in parallel - or at least with GDB (defmt-rtt could maybe be disabled during that time as the messages can be seen when stepping through with the debugger?)

jonathanpallant, 2023-03-08:
:tada:

https://user-images.githubusercontent.com/18005923/223785926-b87c4e43-ef5c-457f-aa5d-37764ea241fb.mov

As it says, I need to find a way to stop the GDB server thread when the user presses Ctrl-C in the `probe-run` window. Should be easy enough.

jonathanpallant, 2023-03-09:
Reading through https://github.com/knurling-rs/probe-run/pull/315, I can see there are some unresolved questions.

* [ ] Should we offer "GDB on Startup", "GDB on Crash", or both?
* [ ] If both, should setting the GDB server port be a different command-line option to enabling the GDB server? We could just use a default port if GDB is enabled but the port isn't specified. It certainly seems nicer to tell people what the port is rather than make them type out `127.0.0.1:3333` or something.
* [ ] Should we patch/fork the probe-rs gdb_server to use a parking_lot::FairMutex to ensure a tight RTT read loop can't steal the connection. Or should we just add a small delay to the RTT loop, like cargo-embed does.

jonathanpallant, 2023-03-09:
I updated #388 to support both GDB on Start and GDB on Fault (borrowing from #315).