knurling-rs/probe-run #91: --features defmt precludes running firmware without it

ianrrees, 2020-11-02:
When probe-run has the defmt feature enabled, using it to run a firmware that uses rtt-target results in `Error: defmt version symbol not found` then probe-run exits.

**To Reproduce**
Steps to reproduce the behavior:

1. Install probe-run with --features defmt
2. Build an ELF that uses RTT, but not defmt
3. `probe-run elf_from_step_2 --chip SOMECHIP`

**Expected and observed behavior**
I'd expect to see either the same behaviour as when probe-run is not built with --features defmt, perhaps with a warning to the effect of "defmt version symbol not found, falling back to raw RTT".  Actually, an error message is displayed and probe-run exits.

**config.toml** 
``` toml
[build]
target = "thumbv6m-none-eabi"

[target.thumbv6m-none-eabi]
runner = "probe-run --chip ATSAMD21G18AU"
rustflags = [
  "-C", "link-arg=-Tlink.x",
]
```

**Probe details**
You can get this information from [`probe-rs-cli`](https://crates.io/crates/probe-rs-cli). Your microcontroller must be connected to your PC / laptop when you run the command below.

``` console
➜  main git:((aba2ddc...)) ✗ probe-run --list-probes
The following devices were found:
[0]: STLink V2-1 (VID: 0483, PID: 374b, Serial: 0670FF505055877267123015, STLink)
```

**Operating System:**
Linux (Ubuntu 20.04)

**ELF file (attachment)**
[codecraft_demo.zip](https://github.com/knurling-rs/probe-run/files/5474181/codecraft_demo.zip)

**Additional context**
I'm totally digging probe-run and defmt, thanks for making and sharing it!

jonas-schievink, 2020-11-02:
Thanks for the report! I've opened https://github.com/knurling-rs/defmt/pull/224 with a fix for this.

jonas-schievink, 2020-11-02:
https://github.com/knurling-rs/probe-run/pull/92 updates defmt to get the fix. @ianrrees can you test if this is now working again?

ianrrees, 2020-11-02:
Thanks for the quick fix @jonas-schievink , it works great!

closed 2020-11-02 by ianrrees