knurling-rs/probe-run #113: Could not compile probe-run v0.1.6 on macOS 10.15.7

andresv, 2020-11-26:
So far I have installed probe-run without problems on macOS (x86_64) with:
```
cargo install --git https://github.com/knurling-rs/probe-run --branch main --features defmt
```

Now I tried to install release version with:
```
cargo install probe-run
```

And got following error:
```
  = note: Undefined symbols for architecture x86_64:
            "_NSAppKitVersionNumber", referenced from:
                _hid_init in libhidapi-6c82a0d52f552184.rlib(hid.o)
                _hid_enumerate in libhidapi-6c82a0d52f552184.rlib(hid.o)
                _hid_open_path in libhidapi-6c82a0d52f552184.rlib(hid.o)
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)

error: aborting due to previous error

error: failed to compile `probe-run v0.1.6 (https://github.com/knurling-rs/probe-run?branch=main#013c0bd3)`, intermediate artifacts can be found at `/var/folders/5b/jlxfj70x1fb_n2l6qh21xxfh0000gn/T/cargo-installcEziMp`

Caused by:
  could not compile `probe-run`.

To learn more, run the command again with --verbose.
```

Rust version:
```
rustc 1.46.0 (04488afe3 2020-08-24)
```

japaric, 2020-11-26:
thanks for the report. I *think* this may have been caused by a recent update in the `hidapi` crate. Could you try to build this crate for me:

``` rust
use hidapi as _;

fn main() {}
```

``` toml
# Cargo.toml
[dependencies]
hidapi = "1.2.4"
```

If that also produces a linker error then this would be a upstream issue in the `hidapi` crate
```

japaric, 2020-11-26:
with help from @justahero we have confirmed that this is an issue in hidapi 1.2.4
this fails to link
``` toml
# Cargo.toml
[dependencies]
hidapi = "1.2.4"
```
this works:
```
# Cargo.toml
[dependencies]
hidapi = "=1.2.3"
```
we'll report the issue upstream in a bit.
I think we can work around the problem in the next patch release of probe-run.

andresv, 2020-11-26:
This was actually successful:
```
   Compiling cc v1.0.65
   Compiling pkg-config v0.3.19
   Compiling libc v0.2.80
   Compiling hidapi v1.2.4
   Compiling hello v0.1.0 (/Users/andres/Downloads/hello)
    Finished dev [unoptimized + debuginfo] target(s) in 2.11s
```

japaric, 2020-11-26:
@andresv was that with `cargo build`? and did you include `use hidapi as _` in the source file? (`cargo check` will not perform linking. omitting the import will not link the library)

japaric, 2020-11-26:
> was that with cargo build? and did you include use hidapi as _ in the source file?

if it still builds and links OK with those, what Rust version are you using? please report `rustc -V`

andresv, 2020-11-26:
`rustc 1.48.0 (7eac88abb 2020-11-16)`

main.rs
```rust
use hidapi as _;

fn main() {

}
```
cargo.rs
```toml
[package]
name = "hello"
version = "0.1.0"
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
#hidapi = "=1.2.3"
hidapi = "=1.2.4"

```

[image]

japaric, 2020-11-26:
strange! maybe `cargo install probe-run` will work for you now @andresv

japaric, 2020-11-26:
if not it could an interaction of different crates what causes the issue ...

andresv, 2020-11-26:
Just tried `cargo install probe-run` again with both 1.46 and 1.48 (i am using 1.46 because `scroll` does not build on later versions) and still got the same error although simple example builds nicely.

andresv, 2020-11-26:
I am searching which commit killed it.

andresv, 2020-11-26:
Whaatt.
Actually if i just do:
```
git clone https://github.com/knurling-rs/probe-run.git
cd probe-run
cargo build
```

Then it works.

japaric, 2020-11-26:
@andresv that's probably because the repo contains a `Cargo.lock` file. when you do `cargo install --git $something` the `Cargo.lock` file included in the checkout is ignored.

`cargo install --git $something --frozen` should also work (I think it's "frozen"; it could be "locked")

japaric, 2020-11-26:
@andresv could you try (see #114)
``` console
$ cargo install --git https://github.com/knurling-rs/probe-run --branch pin-hidapi
```

I'm assuming that `cargo install --git https://github.com/knurling-rs/probe-run --branch main` errors like in the issue description

andresv, 2020-11-26:
Yes same error with
```
cargo install --git https://github.com/knurling-rs/probe-run --branch main
```
however
```
cargo install --git https://github.com/knurling-rs/probe-run --branch pin-hidapi
```
works!

japaric, 2020-11-26:
ok, we'll ship #114 in the next patch release to fix `cargo install probe-run` on macOS

there seems to be *some* issue with hidapi 1.2.4 on macOS but we have not been able to nail down the cause. The minimal repro from https://github.com/knurling-rs/probe-run/issues/113#issuecomment-734188076 failed to link for @justahero in *some* cases but linked fine most of the time.

I'll report the issue in the hidapi repo and see if the hidapi devs have a better idea about what's going on

closed 2020-11-26 by japaric

andresv, 2020-11-27:
Should be fixed in `hidapi 1.2.5`: https://github.com/ruabmbua/hidapi-rs/commit/76e0f904e5151179d0c927b9d06a5600702d388f

PR #114 (2020-11-26): pin hidapi dependency to version 1.2.3
~~unsure if this actually helps or not~~
fixes #113 (work around)