knurling-rs/probe-run #115: Check the selected processor and target for compatability.

tl8roy, 2020-11-27:
After spending an inordinate amount of time trying to work out why RTT wasn't working, I discovered that I had set the wrong target.

As ```probe-run``` knows both the source target (through the path) and the target IC, it should be possible to show a warning if the two aren't compatible with each other.

japaric, 2020-11-30:
This sounds like a good idea to me. I think we may only be able to implement this in a "best effort" manner though rather than as something with 100% accuracy.

Impl notes from the top of my head:

1. look into the `core_type` field of this `probe_rs::Target`

https://github.com/knurling-rs/probe-run/blob/5fe5ed44e96cf555e5a39f62f87bb11dfb200c00/src/main.rs#L127

2. try to extract the compilation target (e.g. `thumbv6m-none-eabi`) from this path. Note that probe-run can be invoked like this: `probe-run ./my-elf`; in that case there's no target information in the path (the `None` case)

https://github.com/knurling-rs/probe-run/blob/5fe5ed44e96cf555e5a39f62f87bb11dfb200c00/src/main.rs#L122

3. compare (1) and (2) and emit a suggestion or not

- for `M0`, target MUST be `thumbv6m-none-eabi`
- for `M3`, target SHOULD be `thumbv7m-none-eabi`
- for `M4` and `M7`, target SHOULD be `thumbv7em-none-eabi`. (`thumbv7m-none-eabi`  may also be OK but probe-rs has no information about FPU support)
- for `Riscv`, I don't think we can give any suggestion as we don't have any info about instruction set support from probe-rs (I vs IMAC vs IMC)

Urhengulas, 2022-04-14:
Which compilation should it be in case of `CoreType::Armv8m` which is Cortex M23 M33?
In the "Platform Support" ([link](https://doc.rust-lang.org/nightly/rustc/platform-support.html)) list I can find `thumbv8m.base-none-eabi`, `thumbv8m.main-none-eabi` and `thumbv8m.main-none-eabihf`. Can it be any of them?

commit 53424597 (2022-04-14): Merge #319

319: Warn on target chip mismatch r=jonas-schievink a=Urhengulas

This PR makes `probe-run` emit a warning, in case the chip type (obtained from `probe-rs`) and the compilation target (obtained from the path to the elf file) do not match. The path will contain the compilation target if `probe-run` is called through `cargo run`. In other cases the path might not contain the compilation target. In these cases we take no action and just move on.

**Example**
```console
$ cargo rb hello -q
(HOST) WARN  Compilation target (thumbv8m.base-none-eabi) and core type (Armv7em) do not match. Y [... cut]
PR #319 (2022-04-14): Warn on target chip mismatch
This PR makes `probe-run` emit a warning, in case the chip type (obtained from `probe-rs`) and the compilation target (obtained from the path to the elf file) do not match. The path will contain the compilation target if `probe-run` is called through `cargo run`. In other cases the path might not contain the compilation target. In these cases we take no action and just move on.

**Example**
```console
$ cargo rb hello -q
(HOST) WARN  Compilation target (thumbv8m.base-none-eabi) and core type (Armv7em) do not match. Your compilation target should be 'thumbv7em-none- [... cut]
closed 2022-04-14 by bors[bot]