knurling-rs/probe-run #124: Stack backtrace is printed after shell prompt

jonas-schievink, 2020-12-17:
```
...
0.000000 INFO  (7/10) running `filter_list32_std`...
└─ integration::tests::__defmt_test_entry @ tests/integration.rs:73
0.000000 INFO  (8/10) running `filter_list32_ext`...
└─ integration::tests::__defmt_test_entry @ tests/integration.rs:73
0.000000 INFO  (9/10) running `dequeue_lower_priority_frame`...
└─ integration::tests::__defmt_test_entry @ tests/integration.rs:73
0.000000 INFO  (10/10) running `ext_roundtrip`...
└─ integration::tests::__defmt_test_entry @ tests/integration.rs:73
^C⏎
~/dev/bxcan/testsuite *master> stack backtrace:
   0: core::ptr::read_volatile
        at /home/jonas/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1042
   1: vcell::VolatileCell<T>::get
        at /home/jonas/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.2/src/lib.rs:32
   2: bxcan::pac::generic::Reg<U,REG>::read
        at /home/jonas/dev/bxcan/src/pac/generic.rs:52
   3: bxcan::Tx<I>::is_idle
        at /home/jonas/dev/bxcan/src/lib.rs:589
   4: bxcan::Can<I>::is_transmitter_idle
        at /home/jonas/dev/bxcan/src/lib.rs:385
   5: integration::tests::ext_roundtrip
        at tests/integration.rs:417
   6: main
        at tests/integration.rs:73
   7: Reset
        at /home/jonas/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.13/src/lib.rs:526
```

Note how the shell prompt shows up between the defmt output and the stacktrace. Not really sure what causes this behavior, but we might need to add a flush somewhere?

Lotterleben, 2021-04-07:
Were these the tests you were running? https://github.com/stm32-rs/bxcan/blob/master/testsuite/tests/integration.rs

jonas-schievink, 2021-04-07:
Yeah