knurling-rs/probe-run #128: WFI/WFE after print results in the same message printed multiple times

PvdBerg1998, 2021-01-02:
I'm implementing an embedded async runtime on an stm32f103. Inside a future I am calling `defmt::info!`, before waiting on a timer (via an interrupt). Therefore, the future returns `Poll::Pending` and I call `asm::wfi()`  to sleep the CPU.
Then, the message printed before the wait gets sent a ton of times, all with the same timestamp. Is this expected? How does your printing mechanism interact with sleep modes?

I've tried setting the `DBGMCU_CR.DEBUG_SLEEP` flag but it doesn't do anything.

This is on Windows 10 with an STLink v2. I don't have a small reproducer at the moment because the code is tied into all sorts of unrelated stuff. Maybe there's an underlying reason this doesn't work and you don't need an example, let me know.

japaric, 2021-01-04:
>  the message printed before the wait gets sent a ton of times,

this could indicate that your executor is continuously polling the future. it could be that some interrupt flag is not being properly cleared and that an interrupt is continuously firing and making the executor wake up.

>  all with the same timestamp

that does sound strange. how are you generating the timestamp? if you are using the atomic counter then you should definitively see a monotonically increasing timestamp.

>  Therefore, the future returns Poll::Pending and I call asm::wfi() to sleep the CPU.

(also, I would suggest using WFE: https://github.com/ferrous-systems/async-on-embedded/issues/1 )

---

>  I don't have a small reproducer 

I would suggest testing sleep behavior with something like shown below using the atomic counter timestamp. We have seen a few bugs with the st-link so it could be an issue in probe-rs.

``` rust
// NOTE untested
#[cortex_m_rt::entry]
fn main() {
    // configure a systick interrupt to fire in 1 second or so
    defmt::info!("A");
    asm::wfe();
    defmt::info!("C");
    app::exit();
}

#[cortex_m_rt::exception]
fn SysTick() {
    defmt::info!("B");
}
```

PvdBerg1998, 2021-01-04:
Yeah I'm using the monotonic timer. My executor is definitely not polling all the time, and even if it were, it wouldn't result in this behaviour: the future is printing the message _before_ the suspension point. I'll get back to you when I've tested some more.

David-OConnor, 2021-01-18:
Same issue, on Stm32L4 and F3.

PvdBerg1998, 2021-01-18:
Sorry I haven't yet had the time to test this.

PvdBerg1998, 2021-01-19:
I got some more info while working on my project. I use `WFI` to sleep the cpu while doing nothing. Just on its own, this causes the debugger to disconnect with a timeout. If I set `dbgmcu.cr.write(|w| w.dbg_sleep().set_bit());`, i.e. the "allow debugging while sleep mode" bit, it doesn't disconnect, but shows this weird behaviour.

Example:
Normal defmt output:
```
..snip
1371.158786 INFO  sleeping...
└─ main::main_task::{{closure}} @ src\bin\main.rs:73
```

Followed by N repeats of the following:

```
   0.000000 INFO  sleeping...
└─ main::main_task::{{closure}} @ src\bin\main.rs:73
```

As you can see there's something weird with the timestamp.
I'm using `DWT::get_cycle_count()` for the timestamp. Could this peripheral be disabled during sleep mode?

jonas-schievink, 2021-01-19:
Hmm, perhaps SRAM writes are disabled in that state? That would result in the RTT read pointer not getting updated properly, so we read the same data again and again.

japaric, 2021-01-20:
> I'm using a global static that stores the SYSCLK (for time calculations). After a while of this weird repeating behaviour, I got a panic that this value was not set. This should be impossible as it's only set once at the start, indicating memory corruption. 

are you using `flip-link` to prevent stack overflows from corrupting memory? apart from stack overflows only `unsafe` code can corrupt memory.

the other thing that comes to mind is caches. if you have the cache enabled then the debugger may be seeing the old, cached value and not the latest value committed by the cortex-m processor.

---

> That would result in the RTT read pointer not getting updated properly, so we read the same data again and again.

yeah, it could be that the device is configured to disable SRAM writes on sleep mode to save power. (on the other hand, the rtt host could keep its copy of the read pointer and sanity check that its value doesn't go backwards but this requires a read pointer that wraps around at `u32::MAX` instead of a "modulo $size" read pointer. I think the rtt protocol uses the modulo variant)

PvdBerg1998, 2021-01-20:
I'm using `flip-link`, yes. There's no (significant) unsafe code, at least in my code. This corruption only happened once with the debugger flag set and the debugger doing this behaviour, so I think it's fair to assume they're correlated.

Dirbaio, 2021-01-21:
I'm also seeing this. RTT doesn't work on stm32f4 on WFE/WFI. The DBGMCU stuff doesn't fix it (it does fix the "brick" where you couldn't flash while CPU is sleeping, but not RTT)

Repro:

```rust
#[entry]
fn main() -> ! {
    let dp = stm32f4xx_hal::stm32::Peripherals::take().unwrap();
    dp.DBGMCU.cr.modify(|_, w| w.dbg_sleep().set_bit());
    dp.DBGMCU.cr.modify(|_, w| w.dbg_stop().set_bit());
    dp.DBGMCU.cr.modify(|_, w| w.dbg_standby().set_bit());

    info!("Hello World!");

    loop {
        cortex_m::asm::wfe();
    }
}
```

This fails with `ERROR Max number of RTT attach retries exceeded.`. If you change the `wfe` with a `nop` it works fine.

japaric, 2021-02-05:
> This fails with ERROR Max number of RTT attach retries exceeded.

Saw this the other day. Using `--connect-under-reset` fixed that "number of retries exceeded" error but didn't fix the RTT messages getting lost.

japaric, 2021-02-05:
FWIW, OpenOCD 0.10's RTT feature doesn't deal with WFI correctly either. Though you at least get a warning.
``` rust
#[rtic::app(device = stm32l4xx_hal::stm32, monotonic = rtic::cyccnt::CYCCNT)]
const APP: () = {
    #[init(spawn = [periodic])]
    fn init(cx: init::Context) {
        let dp = stm32l4xx_hal::stm32::Peripherals::take().unwrap();
        dp.DBGMCU.cr.modify(|_, w| w.dbg_sleep().set_bit());
        dp.DBGMCU.cr.modify(|_, w| w.dbg_stop().set_bit());
        dp.DBGMCU.cr.modify(|_, w| w.dbg_standby().set_bit());

        rtt_init_print!();
        rprintln!("init");
        asm::bkpt();
        cx.spawn.periodic().unwrap();
    }

    // uncomment to make this work; when not defined `idle` defaults to `loop { WFI }`
    // #[idle]
    // fn idle(_: idle::Context) -> ! {
    //     loop {
    //         asm::nop()
    //     }
    // }

    #[task(schedule = [periodic])]
    fn periodic(cx: periodic::Context) {
        rprintln!("periodic");

        cx.schedule
            .periodic(cx.scheduled + 4_000_000.cycles())
            .unwrap();
    }

    extern "C" {
        fn EXTI0();
    }
};
```

``` console
$ cat openocd.gdb
target extended-remote :3333
load
continue
monitor rtt setup 0x20000000 48 "SEGGER RTT"
monitor rtt start
monitor rtt server start 8888 0

$ # after breakpoint run `nc localhost 8888` on another terminal
$ openocd & cargo run
(gdb) continue # RTT TCP port active at this point
Warn : rtt: Up-channel 0 is not active
rtt: Up-channel 0 is not active
Warn : rtt: Up-channel 0 is not active
rtt: Up-channel 0 is not active
Warn : rtt: Up-channel 0 is not active
rtt: Up-channel 0 is not active
```

``` console
$ nc localhost 8888
init
<nothing else is printed afterwards>
```

japaric, 2021-02-05:
Upstream probe-rs issue https://github.com/probe-rs/probe-rs/issues/350 
Firmware workarounds:
- https://github.com/probe-rs/probe-rs/issues/350#issuecomment-740550519 (worked for me w/ the STM32L433)
- https://github.com/probe-rs/probe-rs/issues/350#issuecomment-697180460

Urhengulas, 2021-03-01:
Closing in favor of upstream issue probe-rs/probe-rs#350.

closed 2021-03-01 by Urhengulas