knurling-rs/probe-run #137: Probe-run is writing 0 to flash

David-OConnor, 2021-01-25:
**Describe the bug**
Probe-run writes 0 to flash on STM32

**To Reproduce**
- Write to flash memory in program
- Run (again) using probe run.

Expected: Flash memory stays how you wrote it
What happens: Flash memory is 0ed.

If you exit the probe-run terminal, then reset the MCU, flash memory writes and reads will work as normal, until you run with probe-run again, at which point the data will be wiped.

jonas-schievink, 2021-01-25:
Does this also happen with cargo-flash? If so, this is likely to be a probe-rs bug.

David-OConnor, 2021-01-26:
I don't know - I'll try to figure that out.

Urhengulas, 2021-03-15:
Hi @David-OConnor, did you get to test if the error also occurs with `cargo-flash`?

jonas-schievink, 2021-05-05:
This is very likely to be a probe-rs issue in the end, since we just call its download API to flash the chip, so closing.

closed 2021-05-05 by jonas-schievink