knurling-rs/probe-run #143: probe-run cannot load ELF linked with '-Tdefmt.x -Tlink.x'

japaric, 2021-02-08:
app-template passes the flags '-Tlink.x -Tdefmt.x' to the linker. ELFs linked with those flags work fine.
if you modify app-template and pass the flags in the reverse order '-Tdefmt.x -Tlink.x' the output ELF doesn't work with probe-run.

You get this kind of output:
``` console
$ probe-run --chip nrf52840 bad-elf
  (HOST) INFO  flashing program (0.00 KiB)
Error: No loadable ELF sections were found.
```

The .text section is marked as allocatable in the ELF though:
``` console
$ readelf -S bad-elf
Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .defmt            PROGBITS        00000000 000054 000002 00      0   0  1
  [ 2] .debug_abbrev     PROGBITS        00000000 000056 004589 00      0   0  1
  [ 3] .debug_info       PROGBITS        00000000 0045df 03280b 00      0   0  1
  [ 4] .debug_aranges    PROGBITS        00000000 036df0 003378 00      0   0  8
  [ 5] .debug_ranges     PROGBITS        00000000 03a168 018420 00      0   0  1
  [ 6] .debug_str        PROGBITS        00000000 052588 039ad8 01  MS  0   0  1
  [ 7] .debug_pubnames   PROGBITS        00000000 08c060 00cb53 00      0   0  1
  [ 8] .debug_pubtypes   PROGBITS        00000000 098bb3 005f71 00      0   0  1
  [ 9] .ARM.attributes   ARM_ATTRIBUTES  00000000 09eb24 00003a 00      0   0  1
  [10] .debug_frame      PROGBITS        00000000 09eb60 008bc4 00      0   0  4
  [11] .debug_line       PROGBITS        00000000 0a7724 02ef8a 00      0   0  1
  [12] .debug_loc        PROGBITS        00000000 0d66ae 0002a9 00      0   0  1
  [13] .comment          PROGBITS        00000000 0d6957 00006d 01  MS  0   0  1
  [14] .symtab           SYMTAB          00000000 0d69c4 002ab0 10     16 456  4
  [15] .shstrtab         STRTAB          00000000 0d9474 0000f0 00      0   0  1
  [16] .strtab           STRTAB          00000000 0d9564 0048bd 00      0   0  1
  [17] .vector_table     PROGBITS        00000000 0dde24 000100 00   A  0   0  4
  [18] .text             PROGBITS        00000100 0ddf24 003de4 00  AX  0   0  4
  [19] .rodata           PROGBITS        00003ee4 0e1d10 000c34 00  AM  0   0 16
  [20] .data             PROGBITS        20000000 0e2944 000030 00  WA  0   0  4
  [21] .bss              NOBITS          20000030 0e2974 000008 00  WA  0   0  4
  [22] .uninit           NOBITS          20000038 0e2974 000400 00  WA  0   0  4
```

OpenOCD is able to flash this ELF file.

It was observed that if you link the ELF with '-Tlink.x -Tdefmt.x' the .text and other sections appear first in the above list of section headers (and have small offsets).

It seems that the `object` library is unable to find the .text section in the ELF.
probe-run uses the `object` library to report the size of loadable sections (the "0.00 KiB" part in the above output)
probe-rs also uses the `object` library to locate loadable sections that need to be flashed on the device.

jonathanpallant, 2023-03-08:
The error "No loadable ELF sections were found", is the text associated with [`probe_rs::flashing::download::FileDownloadError::NoLoadableSegments`](https://github.com/probe-rs/probe-rs/blob/f52176c118c8be3cab50f003baad72b500b0b436/probe-rs/src/flashing/download.rs#L77).

This is only created in one place - [`probe_rs::flashing::loader::FlashLoader::load_elf_data`](https://github.com/probe-rs/probe-rs/blob/f52176c118c8be3cab50f003baad72b500b0b436/probe-rs/src/flashing/loader.rs#L159) when `num_sections == 0`.

The function the provides the sections is [`probe_rs::flashing::download::extract_from_elf`](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/flashing/download.rs#L194).

jonathanpallant, 2023-03-08:
The `extract_from_elf` function:

* Iterates through all the segments in the ELF header
* Checks if the segment is not empty, and has segment type `PT_LOAD`
* Iterates through all the sections in the binary
* Checks if they overlap and, if so, adds the segment to the output list

I'll build a test binary and try and flash it but with trace logging enabled.

jonathanpallant, 2023-03-08:
My test program flashed and ran OK. Running `readelf -S` reports:

```text
$ /opt/homebrew/opt/binutils/bin/readelf -S ./target/thumbv7em-none-eabihf/debug/hello 
There are 24 section headers, starting at offset 0x11cd0c:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .defmt            PROGBITS        00000000 001ba0 000009 00      0   0  1
  [ 2] .debug_loc        PROGBITS        00000000 001ba9 0082dc 00      0   0  1
  [ 3] .debug_abbrev     PROGBITS        00000000 009e85 00222f 00      0   0  1
  [ 4] .debug_info       PROGBITS        00000000 00c0b4 03f13a 00      0   0  1
  [ 5] .debug_aranges    PROGBITS        00000000 04b1ee 003598 00      0   0  1
  [ 6] .debug_ranges     PROGBITS        00000000 04e786 01be00 00      0   0  1
  [ 7] .debug_str        PROGBITS        00000000 06a586 057a5a 01  MS  0   0  1
  [ 8] .debug_pubnames   PROGBITS        00000000 0c1fe0 01b34c 00      0   0  1
  [ 9] .debug_pubtypes   PROGBITS        00000000 0dd32c 0078a5 00      0   0  1
  [10] .ARM.attributes   ARM_ATTRIBUTES  00000000 0e4bd1 00003a 00      0   0  1
  [11] .debug_frame      PROGBITS        00000000 0e4c0c 00916c 00      0   0  4
  [12] .debug_line       PROGBITS        00000000 0edd78 02ba87 00      0   0  1
  [13] .comment          PROGBITS        00000000 1197ff 000013 01  MS  0   0  1
  [14] .symtab           SYMTAB          00000000 119814 001390 10     16 162  4
  [15] .shstrtab         STRTAB          00000000 11aba4 0000fd 00      0   0  1
  [16] .strtab           STRTAB          00000000 11aca1 00206a 00      0   0  1
  [17] .vector_table     PROGBITS        00000000 000114 000100 00   A  0   0  4
  [18] .text             PROGBITS        00000100 000214 001500 00  AX  0   0  4
  [19] .rodata           PROGBITS        00001600 001720 00042c 00   A  0   0 16
  [20] .data             PROGBITS        2003fbc0 001b4c 000038 00  WA  0   0  4
  [21] .gnu.sgstubs      PROGBITS        00001a80 001ba0 000000 00  WA  0   0 32
  [22] .bss              NOBITS          2003fbf8 001ba0 000008 00  WA  0   0  4
  [23] .uninit           NOBITS          2003fc00 001ba0 000400 00  WA  0   0  4
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  D (mbind), y (purecode), p (processor specific)
```

So this appears fixed?

jonathanpallant, 2023-03-08:
Just to confirm I reset the link arguments and the sections are in the other order:

```text
➜  test-pointers git:(main) ✗ /opt/homebrew/opt/binutils/bin/readelf -S ./target/thumbv7em-none-eabihf/debug/hello
There are 24 section headers, starting at offset 0x11cd0c:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .vector_table     PROGBITS        00000000 000114 000100 00   A  0   0  4
  [ 2] .text             PROGBITS        00000100 000214 001500 00  AX  0   0  4
  [ 3] .rodata           PROGBITS        00001600 001720 00042c 00   A  0   0 16
  [ 4] .data             PROGBITS        2003fbc0 001b4c 000038 00  WA  0   0  4
  [ 5] .gnu.sgstubs      PROGBITS        00001a80 001ba0 000000 00  WA  0   0 32
  [ 6] .bss              NOBITS          2003fbf8 001ba0 000008 00  WA  0   0  4
  [ 7] .uninit           NOBITS          2003fc00 001ba0 000400 00  WA  0   0  4
  [ 8] .defmt            PROGBITS        00000000 001ba0 000009 00      0   0  1
  [ 9] .debug_loc        PROGBITS        00000000 001ba9 0082dc 00      0   0  1
  [10] .debug_abbrev     PROGBITS        00000000 009e85 00222f 00      0   0  1
  [11] .debug_info       PROGBITS        00000000 00c0b4 03f13a 00      0   0  1
  [12] .debug_aranges    PROGBITS        00000000 04b1ee 003598 00      0   0  1
  [13] .debug_ranges     PROGBITS        00000000 04e786 01be00 00      0   0  1
  [14] .debug_str        PROGBITS        00000000 06a586 057a5a 01  MS  0   0  1
  [15] .debug_pubnames   PROGBITS        00000000 0c1fe0 01b34c 00      0   0  1
  [16] .debug_pubtypes   PROGBITS        00000000 0dd32c 0078a5 00      0   0  1
  [17] .ARM.attributes   ARM_ATTRIBUTES  00000000 0e4bd1 00003a 00      0   0  1
  [18] .debug_frame      PROGBITS        00000000 0e4c0c 00916c 00      0   0  4
  [19] .debug_line       PROGBITS        00000000 0edd78 02ba87 00      0   0  1
  [20] .comment          PROGBITS        00000000 1197ff 000013 01  MS  0   0  1
  [21] .symtab           SYMTAB          00000000 119814 001390 10     23 162  4
  [22] .shstrtab         STRTAB          00000000 11aba4 0000fd 00      0   0  1
  [23] .strtab           STRTAB          00000000 11aca1 00206a 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  D (mbind), y (purecode), p (processor specific)
```

jonathanpallant, 2023-03-08:
@Urhengulas are you happy we close this one?

jonathanpallant, 2023-03-08:
For completeness, here are the *segments*, that is, components of the *Program Header* that are being iterated through in the outer loop in the `extract_from_elf` function:

They are the same in either file:

```console
$ /opt/homebrew/opt/binutils/bin/readelf -l ./target/thumbv7em-none-eabihf/debug/hello

Elf file type is EXEC (Executable file)
Entry point 0x101
There are 6 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x000114 0x00000000 0x00000000 0x00100 0x00100 R   0x4
  LOAD           0x000214 0x00000100 0x00000100 0x01500 0x01500 R E 0x4
  LOAD           0x001720 0x00001600 0x00001600 0x0042c 0x0042c R   0x10
  LOAD           0x001b4c 0x2003fbc0 0x00001a2c 0x00038 0x00038 RW  0x4
  LOAD           0x001ba0 0x2003fbf8 0x2003fbf8 0x00000 0x00408 RW  0x4
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0

 Section to Segment mapping:
  Segment Sections...
   00     .vector_table 
   01     .text 
   02     .rodata 
   03     .data 
   04     .bss .uninit 
   05   
```

Urhengulas, 2023-03-10:
> @Urhengulas are you happy we close this one?

Sure. It seems to be fixed.

closed 2023-03-10 by Urhengulas