knurling-rs/probe-run #144: probe-run on an Arduino Nano 33 BLE?

bryanburgers, 2021-02-11:
**Describe the bug**

I'm not quite sure where to ask this, because I think it's more of a "new to embedded development" question more than anything.

I'm trying to get started programming an [Arduino Nano 33 BLE](https://store.arduino.cc/usa/nano-33-ble) in Rust. It looks like the Nano 33 is a `nRF52840_xxAA`, which seems to be a supported chipset.

I have it hooked up via USB to my Mac and I can flash to it via the Arduino IDE (first new-to-this question as I was digging around trying to understand what's going on: is that because this chip comes pre-flashed with some Arduino bootloader?)

When I do a `cargo run`, I get

```
$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `probe-run --chip nRF52840_xxAA target/thumbv7em-none-eabi/debug/arduino-nano-test`
Error: no probe was found
```

`probe-run --list-probes` tells me that no devices were found. I can't tell if that's _expected_ or if I need to do something else or if I'm completely out in left field.

```
$ probe-run --verbose --list-probes
No devices were found.
$ probe-rs-cli list
No devices were found.
```

I remember a previous co-worker at a previous job explaining things like J-Link to me; does probe-run only work with one of those things? Or should I expect probe-run to work directly connected to the Nano's USB port? Do I need to somehow flash a different bootloader onto the device first (I've seen `avrdude` mentioned...)?

**main.rs**

```rust
#![no_std]
#![no_main]

use cortex_m::asm;
use cortex_m_rt::entry;
use rtt_target::{rprintln, rtt_init_print};
use nrf52840_hal as _;

#[panic_handler]
fn panic(_info: &core::panic::PanicInfo) -> ! {
    loop {
        asm::bkpt()
    }
}

#[entry]
fn main() -> ! {
    rtt_init_print!();
    rprintln!("Hello, world!");

    loop {
        asm::bkpt()
    }
}
```

**Cargo.toml**

```toml
[package]
name = "arduino-nano-test"
version = "0.1.0"
authors = ["Bryan Burgers <bryan@burgers.io>"]
edition = "2018"

[dependencies]
cortex-m = "0.7"
cortex-m-rt = "0.6"
rtt-target = { version = "0.3", features = ["cortex-m"] }
nrf52840-hal = "0.12"
```

**config.toml**
The contents of your project's `.cargo/config.toml` file

``` toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip nRF52840_xxAA"
rustflags = [
  "-C", "link-arg=-Tlink.x",
]

[build]
# target = "thumbv7em-none-eabihf"  # I think it's an hf, but also saw somewhere that hf wasn't supported and need to workaround without it, but maybe that was old news
target = "thumbv7em-none-eabi"
```

**Probe details**

``` console
$ probe-rs-cli list
No devices were found
```

**Operating System:**
macOS 11.2 (Big Sur)

```
$ uname -ap
Darwin Bryans-MacBook-Pro.local 20.3.0 Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64 x86_64 i386
```

japaric, 2021-02-11:
Hello!

To use `probe-run` you need a "probe" (also known as "debugger") that sits between your PC and the microcontroller (the nRF52840 in your case). Most development boards, specially the bigger ones, have a probe "on-board" so after you connect your PC to the dev board using a USB cable you are all set and can use `probe-run`.

This particular board (the Arduino Nano 33 BLE) does *not* come with an on-board probe. When that's the case you'll need to connect a "stand alone" probe (additional piece of hardware) to the microcontroller and then connect the probe to your PC via USB. Some examples of stand alone probes are: the ST-Link and the J-Link. (There are also CMSIS-DAP based probes but I'm not aware of any stand alone probe that adheres to the CMSIS DAP standard)

Looking at the schematic of this board: https://content.arduino.cc/assets/NANO33BLE_V2.0_sch.pdf
There are SWD pins (SWDCLK and SWDIO) routed to connector J3. You'll need to connect those SWD pins and GND to the standalone probe to be able to use `probe-run`.

Now, I don't see "J3" printed on the board when looking at the pictures of the board. J3 may or may not be the 4 square solder pads on the back of the board that sit between "D6" and "D9".

I would ask on the Arduino forums how one can connect a stand alone debugger to this board. Or if they have some accessory for this.
If they say that the only way to load a new program on the board is using a "bootloader" then this board is not supported by `probe-run` at the moment and would fall in the same group as the Teensy boards (see #22).

Hope that helps!

bryanburgers, 2021-02-11:
@japaric Yes, that is actually _very_ helpful, and helps me understand the difference between "I'm doing this wrong" and "My board doesn't support this".

I really appreciate you taking the time to look into this and help me understand what's going on.

I don't think my hobbying quite merits buying an ST-Link or J-Link and soldering something onto my boards, so I'll look into different ways to start experimenting (probably `avrdude` somehow).

Thanks again for your time! I really appreciate it.

closed 2021-02-11 by bryanburgers