knurling-rs/probe-run #168: defmt-test: hide location info lines that point to the `#[tests]` attribute

japaric, 2021-03-04:
given this test file

``` rust
#[defmt_test::tests] // <- line 8
mod tests {
    #[test]
    fn a() {
        trace!("A");
    }

    #[test]
    fn b() {
        trace!("B");
    }
}
```

`probe-run` output looks like this:

``` console
$ cargo t
0.000000 INFO  (1/2) running `a`...
└─ foo::tests::__defmt_test_entry @ tests/foo.rs:8
0.000000 TRACE A
└─ foo::tests::a @ tests/foo.rs:12
0.000000 INFO  (2/2) running `b`...
└─ foo::tests::__defmt_test_entry @ tests/foo.rs:8
0.000000 TRACE B
└─ foo::tests::b @ tests/foo.rs:17
0.000000 INFO  all tests passed!
└─ foo::tests::__defmt_test_entry @ tests/foo.rs:8
```

the location info shown on the "running" lines is misleading because it points to the `#[tests]` attribute. I think it would be better if no location info was shown on those:

``` console
$ cargo t
0.000000 INFO  (1/2) running `a`...
0.000000 TRACE A
└─ foo::tests::a @ tests/foo.rs:12
0.000000 INFO  (2/2) running `b`...
0.000000 TRACE B
└─ foo::tests::b @ tests/foo.rs:17
0.000000 INFO  all tests passed!
└─ foo::tests::__defmt_test_entry @ tests/foo.rs:8
```

if the "running" lines had custom tags (as suggested in #167) then I think this could be implemented in `probe-run`.

I guess the other option is to crate a set of (private?) `defmt` logging macros that do *not* include location info and then use those from the `defmt-test` proc macro.

jonas-schievink, 2021-03-04:
Couldn't we use the span of the test function name instead of the call-site span to fix this?

japaric, 2021-03-04:
> Couldn't we use the span of the test function name instead of the call-site span to fix this?

actually, that would be even better