knurling-rs/probe-run #169: Add a way to see probe-rs logs.

Dirbaio, 2021-03-04:
There's no way to see logs from probe-rs.

- `--verbose` doesn't work
- `RUST_LOG=trace` doesn't work.

This is a blocker in helping people coming to the probe-rs matrix channel asking for help with flashing issues etc. There's no way for them to provide the logs we need without patching and recompiling probe-run.

This is due to this check here https://github.com/knurling-rs/probe-run/blob/ca0b3c53e28bf607a9db03c01798778a17a967c8/src/main.rs#L117-L121

closed 2021-03-08 by bors[bot]

Urhengulas, 2021-03-08:
Hi @Dirbaio, using #171 you should be able to get all logs by passing `-vv` to `probe-run`. Does this match your expactations?

Dirbaio, 2021-03-08:
Yes! Works great, thank you! :)

commit 590bb2cf (2021-03-08): Merge #171

171: Introduce even more verbose log level r=jonas-schievink a=Urhengulas

Fixes #169
PR #171 (2021-03-08): Introduce even more verbose log level
Fixes #169