knurling-rs/probe-run #170: Fails to decode data after long time running

MathiasKoch, 2021-03-05:
I have just updated both defmt and probe-run, and i am now starting to see a lot of 
```
(HOST) ERROR failed to decode defmt data: [af, 6e, 39, 0, 0, ...]
Error: malformed data
```

They are not coming right away, but rather after running for a long time. I am not logging any significant amount of data, and it doesn't seem like it's a particular log frame that causes the error.

This is kind of a big issue, as above error also seems to halt the core, causing not only our logging, but also our application to stop running.

Would it be possible in cases of malformed data, to simply rotate a single byte and try decoding again, in a loop? It seems to work as a last resort re-synchronization mechanism, that is MUCH more desireable than a panic. That way only the malformed frame is lost.
The obvious alternative would be to finally add framing to the wire format, which i think would also solve a bunch of other issues/requests?

jonas-schievink, 2021-03-05:
Are you using defmt-rtt? Does this happen with any executable or are specific messages etc. needed to reproduce this?

MathiasKoch, 2021-03-05:
Yea, using defmt-rtt.

I can't seem to find the trigger.. It seems like it's random and can happen right after flashing, or 24 hours into the program, so it doesn't seem like its any particular place in the execution. 

Is there anything i can enable to obtain more info?

Urhengulas, 2021-03-14:
Hi @MathiasKoch 👋🏾 

I ran a small sample program (see below) for a couple of hours, but couldn't find the error you described.

Can you please elaborate a bit on your defmt usage and rough setup? What kind of data structures are you logging which produce this error?

Also can you please try to run `probe-run` with `--verbose` enabled. Maybe we can find something odd there.

Thank you!

<details>
<summary>sample code</summary>

```rust
#![no_main]
#![no_std]

use cortex_m::asm;
use defmt::Format; // <- derive attribute
use defmt_app as _; // global logger + panicking-behavior + memory layout

#[derive(Format)]
struct S1 {
    x: u8,
}

#[cortex_m_rt::entry]
fn main() -> ! {
    let s = S1 { x: 42 };
    let x = 43;

    loop {
        defmt::info!("s={:?}", s);
        defmt::info!("x={=u8}", x);
        asm::delay(25_000_000)
    }
}
```
</details>

Urhengulas, 2021-03-14:
Just skimming other issues I assume this is related to https://github.com/knurling-rs/defmt/issues/322

David-OConnor, 2021-03-15:
@MathiasKoch Same experience her, including the apparent "randomness". Ie sometimes right after flashing, sometimes after a few seconds or minds, sometimes after hours.

MathiasKoch, 2021-12-22:
I think this has been solved by rzcobs encoding in v0.3.0?

closed 2021-12-22 by MathiasKoch

Urhengulas, 2022-01-03:
That's good to hear @MathiasKoch!

Do you see data getting lost and defmt recovers the next frame, or is no data getting lost at all? I am trying to understand if the problem of "randomly malformed data after a long time" is solved or if it is just mitigated due to the recovery.

PS: Happy new year! 🎉

MathiasKoch, 2022-01-03:
I am not sure yet, as we have not had time to upgrade our devices to v0.3.0 yet, so you might be right that this is still not fixed. Not sure?

Happy new year! :tada:

Urhengulas, 2022-01-04:
I'd be happy if you can report back in either case. If the root cause is fixed as well if it is not :)