knurling-rs/probe-run #172: Building probe-run against specific defmt

ianrrees, 2021-03-09:
A couple times, I've found it necessary to build my own probe-run (most recently, to use a newer probe-rs), but doing that leads to a defmt version mismatch problem that can be difficult to resolve.  Is there a way to build probe-run against specific version of defmt?

Specifically, if I do `cargo install probe-run`, I currently get probe-run 0.2.1, with supported defmt version 0.2, however doing `cargo install --path .` in probe-run source checked out at v0.2.1 yields a probe-run with supported defmt version 60c6447f8ecbc4ff023378ba6905bcd0de1e679f .

I've tried using Cargo's patch section to update my firmware's defmt version from only one place, but haven't managed to get that working.

Urhengulas, 2021-03-14:
Hi @ianrrees, Can you please check if #174 is a viable workaround for you?

ianrrees, 2021-03-14:
Yep, works great, thanks!

closed 2021-03-14 by ianrrees