knurling-rs/probe-run #176: STM32WB55RGVx gives no apporpriate flash algorithm

tacdom, 2021-03-10:
**Describe the bug**
Creating a simple hello world from the app-template for the STM32WB55 Nucleuo board gives the following error message when running probe run:
`Trying to write flash, but no suitable flash loader algorithm is linked to the given target information.`
As a probe I am using the onboard ST-Link on the nucleo

**To Reproduce**
Steps to reproduce the behavior:
Connect to a STM32WB55 Nucleo and run 'probe-run --chip STM32WB55RGVx <elf>` or run `cargo run --bin <bin name>`

**Expected and observed behavior**
I expected to have my elf flashed and run on the WB55 but get this error:
`Trying to write flash, but no suitable flash loader algorithm is linked to the given target information.`

``` console
$ cargo run --bin hello
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `probe-run --chip STM32WB55RGVx target/thumbv7em-none-eabihf/debug/hello`
  (HOST) INFO  flashing program (6.93 KiB)
Error: Error while flashing

Caused by:
    Trying to write flash, but no suitable flash loader algorithm is linked to the given target information.
```

**config.toml**
The contents of your project's `.cargo/config.toml` file

*Example*:

``` toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip STM32WB55RGVx"
rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C", "link-arg=--nmagic",
]

[build]
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)

[alias]
rb = "run --bin"
rrb = "run --release --bin"
```

**Operating System:**
MAC OS Catalina and Linux Ubuntu 20

**Additional context**
In my opinion this error is caused by the probe-rs issue [466](https://github.com/probe-rs/probe-rs/issues/466) which is fixed in the current mainline development. Maybe updating to the latest probe-rs will fix the issue

Urhengulas, 2021-03-16:
Hi @tacdom, 
As you correctly stated this issue is already fixed in `probe-rs`. We just need to wait until they release a new version including this change.

japaric, 2022-03-25:
this chip is in the list of probe-run v0.3.3
``` console
$ probe-run --list-chips | rg -i stm32wb55rgvx
        STM32WB55RGVx
```

closed 2022-03-25 by japaric