knurling-rs/probe-run #180: Refactor: shorten & un-nest functions, re-think return values

Lotterleben, 2021-03-31:
Opening this issue for visibility (and also because I'd like to do this :D)
Imo it'd help with readability and maintainability to do a refactoring run.

Things that I've been bumping into:
- [ ] checks in `notmain()` could be pulled into separate functions for readability 
- [ ] `construct_backtrace()` is pretty long and involved; amount of parameters is smelly
- [ ] `construct_backtrace()` always returns OK, even if there was an exception (then it's an `OK(Some(TopException))`). Returning a Result<(), TopException> would be more intuitive imo, but this could also be that I'm misunderstanding `anyhow::Error`
- [ ] `update_cfa()`has another unintuive return value, could be `Option<new_cfa>` imo

These can also be sneaked in as drive-bys on bugfixes and feature implementation.

Urhengulas, 2021-03-31:
Totally agree that there is room for quite some refactoring!

Also we might want to add some unit, but also end-to-end tests. Maybe the kind of @jonas-schievink described in the github-actions blog Post :)

Lotterleben, 2021-04-01:
very much agreed! I've tracked that one in https://github.com/knurling-rs/probe-run/issues/173 :D Maybe we can chat about planning/funds for this in the next bi-weekly?

japaric, 2021-05-12:
`construct_backtrace` was refactored in PR #197

Urhengulas, 2021-07-05:
@Lotterleben Do you consider this resolved by #222?

Lotterleben, 2021-07-06:
very much so! thanks for the bump

closed 2021-07-06 by Lotterleben