knurling-rs/probe-run #182: Why is `PROBE_RUN_IGNORE_VERSION` only checked at compile time?

MathiasKoch, 2021-04-20:
I am not sure i understand why `PROBE_RUN_IGNORE_VERSION` is only checked at compile time? I think it would make sense to allow users to specify it at runtime, as it is quite obvious that it might break stuff? 

https://github.com/knurling-rs/probe-run/blob/df187189849c35cd00e50f2c2146d878c1636de1/src/main.rs#L191

Urhengulas, 2021-04-20:
Hi @MathiasKoch, the rationale is that we don't want users to use it on runtime, but only have it as a workaround for people doing development on `probe-run`. An end user should resolve the version mismatch as specified by the error message.

MathiasKoch, 2021-04-22:
Fair enough, guess i'll have to keep using a locally compiled version for development :/

Urhengulas, 2021-04-25:
> Fair enough, guess i'll have to keep using a locally compiled version for development :/ 

Hi @MathiasKoch, Can you please describe your setup, tell why it doesnt work with the current solution and outline the options/workarounds you see? 
If we see the current solution doesn't make sense in some cases we can definitely adapt it.

MathiasKoch, 2021-04-26:
@Urhengulas 
It's not that it doesn't work with the current setup, but meerly the annoyance of constantly having to `cargo install probe-run --force` & `cargo install --path . --force` in order to do local development with a git/local defmt version, while still flashing/building production builds every day (where we want to make sure it works with the released version)

Allowing the ignore version flag at run-time, would just allow me to only use the crates.io version, as long as there are no breaking changes in defmt

Urhengulas, 2021-04-26:
I see. Checking the variable on runtime is probably not as bad as long as we send a warning that the version check is disabled.

Urhengulas, 2021-05-05:
@MathiasKoch: If you want you can go forward and implement it 😄

commit 1fbda3a8 (2021-05-10): Merge #193

193: Check `PROBE_RUN_IGNORE_VERSION` on runtime r=jonas-schievink a=Urhengulas

Fixes #182
PR #193 (2021-05-10): Check `PROBE_RUN_IGNORE_VERSION` on runtime
Fixes #182
closed 2021-05-10 by bors[bot]