knurling-rs/probe-run #186: investigate if stack unwinding is possible in presence of stack overflow

japaric, 2021-04-29:
as described in PR #184 

> When a program has overflowed its stack, the HardFault exception is called. The exception entry
pushed some registers onto the stack (see Stacked struct). Because there's no or little stack memory
in the stack overflow scenario the registers are pushed outside the RAM region; this results in
register information being lost. Without this information we cannot locate the frame preempted by
HardFault and thus cannot unwind the stack.

in a stack overflow condition we may not have information about the frame preempted by `HardFault`.

It might, however, be possible to locate the previous frame by doing a liner search of the stack from the lower bound of the RAM region. This issue is to investigate whether that's feasible at all or not and how to perform the search.

Urhengulas, 2023-02-11:
@japaric How can the previous frame be identified?

japaric, 2023-02-23:
I don't know. doing the research is part of the task. it could be that's it's not possible at all.

on exception entry the registers are pushed onto the stack in a [well-defined layout](https://docs.rs/cortex-m-rt/0.7.3/src/cortex_m_rt/lib.rs.html#852-861) so by scanning the ram memory one may be able to figure that a word is the program counter (PC) and the one next to it is the link register (LR). however, for regular function calls I don't think the stack pointer and frame pointer (?) values are pushed onto the stack in any standardized way by the function prologue so if the program didn't hit any exception before the stack overflow it may be hard to find a stack frame boundary via scanning. also, whether unwinding is possible to do from *any* frame, rather than from the topmost one, at all is yet another question.

it could be that in this case of stack overflow, the best thing to do in this case is to drop the user into a `(gdb)` shell and let them manually examine the memory. they may have better luck at getting clues out of the GDB as to why the program hit an stack overflow.