knurling-rs/probe-run #198: [RFC] turn `--max-backtrace-len=0` into a CLI parsing error

japaric, 2021-05-12:
(this is technically a breaking change so it should wait until v0.3.0)

`--max-backtrace-len=0` is currently allowed. While working on PR #197 it wasn't clear to me what the semantics of `--force-backtrace --max-backtrace-len=0` should be but I made `--max-backtrace-len=0` never print a backtrace regardless of other flags and the presence of stack overflows or exceptions.

IMO, it'd be better to reject `--max-backtrace-len=0` and only accepts lengths equal or greater than 1; that avoids the above ambiguity.

cc @Lotterleben

Lotterleben, 2021-05-20:
makes sense. I'm wondering, is there a case where one might not want any backtraces ever?

japaric, 2021-05-20:
> I'm wondering, is there a case where one might not want any backtraces ever?

we also print the backtrace to stdout, right? perhaps if you want to grep for something in the logs but *not* in the backtrace then it may be useful to completely disable the backtrace. this use case would be best served by the proposed JSON output (#178)

jonas-schievink, 2021-06-16:
IMO it makes sense to reject `--force-backtrace --max-backtrace-len=0`, but I don't think `--max-backtrace-len=0` by itself should be rejected. I would expect it to just turn off the backtrace (either completely, with no output referring to the backtrace, or in a way that just prints a backtrace with 0 frames).

japaric, 2021-07-07:
> but I don't think --max-backtrace-len=0 by itself should be rejected. I would expect it to just turn off the backtrace

that sounds like 'some way to disable the backtrace' is a desirable thing.

perhaps we can harmonize this to work like Cargo's `--color` flag? `--backtrace={auto,always,never}`. `auto` is the current, default behavior (e.g. backtrace on panic, but not on `exit`). `always` is `--force-backtrace` (which is no longer needed with this scheme). `never` disables the backtrace completely.

THEN `--max-backtrace-len` can be limited to be greater than 1 and shall only be used with `--backtrace={auto,always}`. e.g. `--max-backtrace-len=10 --backtrace=never` would be an error.

OR we don't restrict mixing `--max-backtrace-len` and `--backtrace=never` (i.e. that's not an error) but simply ignore the `--max-backtrace-len` flag in that case.

---

Also, I'd like the default to be "no limit on backtrace length" (e.g. if the backtrace is the result of a hard to reproduce condition then you want *all the info you can get your hands on*; not just some arbitrary number of frames). Right now, the backtrace is limited to 50 frames and there's no good way to undo it -- I mean you can use `--max-backtrace-len=999` or something like that but that looks a bit ridiculous. 

We could make the default be 'the number of backtrace frames is not limited' OR we could make `--max-backtrace-len=0` mean 'do not limit the backtrace size' *and* keep the max-backtrace-len default of 50 frames. I'd prefer the former option.

jonas-schievink, 2021-07-07:
We could also rename it to `--backtrace-limit`, and have 0 mean "unlimited", I believe that's what some tools do

japaric, 2021-07-29:
OK, let me more concretely propose this:

- `--force-backtrace` is removed
- `--backtrace-len` is renamed to `--backtrace-limit`
- `--backtrace` is added

The interaction between these flags is specified in the table below:

| `--backtrace=` | `--backtrace-limit=N` | Backtrace is printed                  |
|----------------|-----------------------|---------------------------------------|
| `never`        | is ignored            | never                                 |
| `auto`         | has an effect         | iff program panics or stack overflows |
| `always`       | has an effect         | always                                |

Additionally,

- `--backtrace` flag is optional and defaults to `auto`
- `--backtrace-limit` flag is optional and defaults to `0` (+)
- `--backtrace-limit=0` is accepted and means "no limit"

(+) rationale on `0` vs current `50`:
1. backtrace information is invaluable for debugging issues and the panic condition may be hard to trigger (because interactions with the environment) so discarding that information with a default `--limit` could waste a lot of time in reproducing the panic condition.
2. an infinite backtrace is a bug in `probe-run` and it should be promptly fixed, instead of being papered over with an artificial `--limit`

@jonas-schievink , @Lotterleben can I get a 'sounds good to me' / 'I have a concern' reaction on the above proposal as **a comment** -- I don't get notified about emoji reactions to comments so that would be easy to miss.

jonas-schievink, 2021-07-29:
> an infinite backtrace is a bug in `probe-run` and it should be promptly fixed, instead of being papered over with an artificial `--limit`

I agree in principle, but we've seen programs that even resulted in an infinite backtrace in GDB, so I don't think this is realistic (https://github.com/knurling-rs/probe-run/issues/127#issuecomment-810324552).

Everything else sounds good to me though!

japaric, 2021-07-29:
Fair, I can compromise on keeping the `50` frames limit.

Lotterleben, 2021-07-29:
+1 to what Jonas says. We also picked the 50 line limit quite arbitrarily, so I'd be OK with extending it.

japaric, 2021-08-02:
OK. That's two approvals. I'll amend my proposal; link it from the issue description as accepted and change this ticket status to 'needs PR'

closed 2021-09-02 by bors[bot]