knurling-rs/probe-run #205: figure-out & document how to make "path remapping" work reliably

japaric, 2021-05-18:
in backtraces and log messages (?) the source code location of core / std API points to a dummy `/rustc/` folder. Example below:

``` console
$ cargo rb panic
(..)
   5: core::panicking::panic_fmt
        at /rustc/9bc8c42bb2f19e745a63f3445f1ac248fb015e53/library/core/src/panicking.rs:92:14
```

in some instances this path is automatically converted to a local path that points into the `rust-src` component. This has been seen in the issue description of #201, for example.

``` console
(..)
   6: core::panicking::panic
        at /rustc/881c1ac408d93bb7adaa3a51dabab9266e82eee8/library/core/src/panicking.rs:50:5
(..)
  12: core::sync::atomic::atomic_load
        at /home/jonas/.rustup/toolchains/nightly-2021-05-09-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:2354:24
```

It's unclear how to make this remapping mechanism work reliably but figuring that out and documenting it in `probe-run` docs would effectively fix #66

It's also unclear how the remapping works: Is the "`/rustc/` path must be remapped" info stored in DWARF (debug info)? Where does the remapping actually happen? in the `gimli` crate? or in the `addr2line` crate?

Urhengulas, 2021-06-28:
Do we still desire to document this, although both are shortened by now?

japaric, 2021-06-29:
Yes, I think it's still worthwhile to do because the `/rustc/<hash>` would become `[stable]` when shortened -- right now it just says `[rust]`. So this would help with correctly identifying the toolchain and displaying its name in the output. 

When *not* shortening the paths this would be even more helpful because it would be possible to ctrl+click the (remapped) absolute path to open the file in VS code. ctrl+click-ing the fake `/rustc/<hash>` doesn't do anything in VS code right now.

Though I agree this is low priority

Urhengulas, 2021-06-29:
> Yes, I think it's still worthwhile to do because the `/rustc/<hash>` would become `[stable]` when shortened -- right now it just says `[rust]`. So this would help with correctly identifying the [...]

Sounds good.

> Though I agree this is low priority

About the "low priority": Everything where I have not really have an option priority-wise gets a "low". Please always feel free to change it. (Similar for difficulty: If I don't know it gets a "medium").

japaric, 2021-06-29:
(assigning difficulty: medium because this requires some research but there are some leads about where to look at)