knurling-rs/probe-run #208: Is there a way to use probe-run just to flash?

azerupi, 2021-05-19:
Hi,

At the moment when I flash the firmware of a device with probe-run it tries to read RTT logs. But my application doesn't use them and I essentially use probe-run just to flash the device.

```
probe-run --chip stm32l0 target/thumbv6m-none-eabi/release/loki-typhon
  (HOST) INFO  flashing program (18.71 KiB)
  (HOST) INFO  success!
RTT logs not available; blocking until the device halts..
```

Is there a way to exit after flashing? I would like to be able to use probe-run in a hardware testbench. I need to flash the device and then perform a couple of measurements. So I would need probe-run to exit after flashing instead of waiting forever. 

If that option does not exist, would it be conceivable to add it?

jonas-schievink, 2021-05-19:
This use case is normally covered by https://github.com/probe-rs/cargo-flash, is there a reason that doesn't work for you?

azerupi, 2021-05-19:
I might be mistaken, but cargo-flash seems to be centered around a crate and cargo.
For my use case I have already built the firmware and produced an elf file (or hex file) and now I would like to flash that on multiple chips.

I think probe-run is quite easy to use and so I would have liked to base my solution around it, but I can totally understand if this feature falls out of the scope of the project.

Urhengulas, 2021-05-19:
As far as I see you can optionally specify the `elf` using `cargo-flash` (see `cargo flash --help` and https://github.com/probe-rs/cargo-flash/blob/master/src/main.rs#L107-L112).

Going to close this, but feel free to re-open in case you have further questions 😄

closed 2021-05-19 by Urhengulas