knurling-rs/probe-run #213: probe-run doesn't print a backtrace on UDF

jonathanpallant, 2021-05-21:
**Describe the bug**
I used probe-run to run my program on an STM32F767-Nucleo. When the program hit an `asm::udf()` call, probe-run did not print a backtrace and instead told me I had debug information.

I checked in GDB and I do have symbols.

**To Reproduce**

1. Put an "asm::udf()" in your panic handler.
2. Set your program to panic automatically.
3. Build with the debug profile with `cargo build`
4. Start program with `probe-run --chip=STM32F767ZITx ./target/thumbv7em-none-eabihf/debug/test-app`
5. Trigger the crash.

**Expected and observed behavior**
I expected to see a backtrace but instead it told me I didn't have any symbols.
``` console
Panic! PanicInfo { payload: Any, message: Some(Manual panic was triggered), location: Location { file: "src/main.rs", line: 119, col: 5 } }
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
Error: debug information is missing. Likely fixes:
1. compile the Rust code with `debug = 1` or higher. This is configured in the `profile.{release,bench}` sections of Cargo.toml (`profile.{dev,test}` default to `debug = 2`)
2. use a recent version of the `cortex-m` crates (e.g. cortex-m 0.6.3 or newer). Check versions in Cargo.lock
3. if linking to C code, compile the C code with the `-g` flag

Caused by:
    Do not have unwind info for the given address.
``````

**config.toml**
The contents of your project's `.cargo/config.toml` file

``` toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip=STM32F767ZITx"

rustflags = [
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C",
  "link-arg=--nmagic", # LLD (shipped with the Rust toolchain) is used as the default linker
  "-C",
  "link-arg=-Tlink.x", # if you run into problems with LLD switch to the GNU linker by commenting out
  # this line
  # "-C", "linker=arm-none-eabi-ld",
  # if you need to link to pre-compiled C libraries provided by a C toolchain
  # use GCC as the linker by commenting out both lines above and then
  # uncommenting the three lines below
  # "-C", "linker=arm-none-eabi-gcc",
  # "-C", "link-arg=-Wl,-Tlink.x",
  # "-C", "link-arg=-nostartfiles",
]

[build]
# Cortex-M4F and Cortex-M7F (with FPU)
target = "thumbv7em-none-eabihf"
```

**Probe details**

STM32F767-Nucleo

``` console
$ probe-rs-cli list
The following devices were found:
[0]: STLink V2-1 (VID: 0483, PID: 374b, Serial: 066AFF5049558575xxxxxxxx, STLink)
```

**Operating System:**
Ubuntu Linux 20.04

**ELF file (attachment)**

Redacted. AMA.

**Additional context**
Was in a VS Code terminal, connected via SSH to a VM, which has the ST-Link mapped in as a USB device.

jonathanpallant, 2021-05-21:
As a test I ran `probe-rs-cli`, and it was able to show a backtrace:

```console
$ probe-rs-cli debug --exe ./target/thumbv7em-none-eabihf/debug/test-app -c STM32F767ZIT
>> halt
Core stopped at address 0x0800b424
0x0800b424: 00000000ff
0x0800b425: 00000000e7
0x0800b426: 0000000004
0x0800b427: 00000000b0
0x0800b428: 0000000080
0x0800b429: 00000000bd
0x0800b42a: 0000000080
0x0800b42b: 00000000b5
0x0800b42c: 000000006f
0x0800b42d: 0000000046
0x0800b42e: 0000000086
0x0800b42f: 00000000b0
0x0800b430: 0000000004
0x0800b431: 0000000091
0x0800b432: 0000000005
0x0800b433: 0000000090
0x0800b434: 00000000ff
0x0800b435: 00000000e7
0x0800b436: 0000000004
0x0800b437: 0000000099
0x0800b438: 0000000005
0x0800b439: 0000000098
0x0800b43a: 0000000088
0x0800b43b: 0000000042
0x0800b43c: 0000000002
0x0800b43d: 00000000d3
0x0800b43e: 00000000ff
0x0800b43f: 00000000e7
0x0800b440: 0000000006
0x0800b441: 00000000b0
0x0800b442: 0000000080
0x0800b443: 00000000bd
>> bt
0: compiler_fence
        /home/jpallant/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:2746:23
        Variables:

1: HardFault_
        /home/jpallant/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.14/src/lib.rs:563:9
        Variables:

>> reset
>> bt
0: Reset
        /home/jpallant/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.14/src/lib.rs:497
        Variables:

>> run
>> halt
Core stopped at address 0x0800450e
0x0800450e: 000000008d
0x0800450f: 00000000f8
0x08004510: 00000000ad
0x08004511: 0000000011
0x08004512: 000000008d
0x08004513: 00000000f8
0x08004514: 00000000ac
0x08004515: 0000000001
0x08004516: 00000000ff
0x08004517: 00000000e7
0x08004518: 000000009d
0x08004519: 00000000f8
0x0800451a: 00000000ac
0x0800451b: 0000000001
0x0800451c: 00000000c0
0x0800451d: 0000000007
0x0800451e: 0000000008
0x0800451f: 00000000b1
0x08004520: 00000000ff
0x08004521: 00000000e7
0x08004522: 00000000f1
0x08004523: 00000000e7
0x08004524: 000000009d
0x08004525: 00000000f8
0x08004526: 00000000ad
0x08004527: 0000000011
0x08004528: 0000000054
0x08004529: 00000000a8
0x0800452a: 00000000fc
0x0800452b: 00000000f7
0x0800452c: 0000000054
0x0800452d: 00000000f8
>> bt
0: __cortex_m_rt_main
        src/main.rs:168:24
        Variables:

1: __cortex_m_rt_main_trampoline
        src/main.rs:133:1
        Variables:

2: trampoline
        /home/jpallant/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.14/src/lib.rs:547:26
        Variables:

3: Reset
        /home/jpallant/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.14/src/lib.rs:550:13
        Variables:

>> 
```

jonathanpallant, 2022-03-29:
I no longer have access to that STM32F767 board so happy to close.

japaric, 2022-03-29:
let's close it then. if it pops up again feel free to open a new ticket

closed 2022-03-29 by japaric