knurling-rs/probe-run #236: Add option to verify flash

MathiasKoch, 2021-06-29:
It would be nice if you could add a cli option to verify the written flash, as this is now possible with probe-rs 0.11.0, but it default to false.

See https://github.com/probe-rs/probe-rs/pull/671/files for more details.

Suyash458, 2022-07-15:
@Urhengulas I'd like to pick this one up! However, I don't have the hardware set up to verify it

Urhengulas, 2022-07-20:
Hi @Suyash458, Thank you for your interest!

Do you have any microcontroller at hand? All of stm32, nordic, esp32 should work.
But even if not, I am able to test your PR.

Suyash458, 2022-07-22:
@Urhengulas Thanks! I have an esp32 with me so I can test it out, I assume I don't need a JTAG debugger as well?

Urhengulas, 2022-07-25:
@Suyash458 said:
> @Urhengulas Thanks! I have an esp32 with me so I can test it out, I assume I don't need a JTAG debugger as well?

You will need one a ST-Link, J-Link or CMSIS-DAP based probe, since that is what probe-rs supports. See https://probe.rs/docs/getting-started/probe-setup/.

In case you don't have one of those and can't get one I can test it out on real hardware for you.

commit 422d4a9b (2022-09-19): Merge #353

353: Add option to verify written flash r=Urhengulas a=Suyash458

 - add option to verify written flash
 - fixes #236
commit f5dd134a (2022-09-28): Merge #353

353: Add option to verify written flash r=Suyash458 a=Suyash458

 - add option to verify written flash
 - fixes #236
PR #353 (2022-09-28): Add option to verify written flash
 - add option to verify written flash
 - fixes #236
closed 2022-09-28 by bors[bot]