knurling-rs/probe-run #239: add troubleshooting info to "probe not found" error

japaric, 2021-07-09:
`probe-run` could print the info I posted in https://github.com/knurling-rs/probe-run/issues/238#issuecomment-877260800 so users can troubleshoot / debug the issue on their own

Urhengulas, 2021-07-09:
We can additionally add it to the Troubleshooting section of the README. And maybe put it to a separate file, since it got quite long by now.

japaric, 2021-07-12:
>  the Troubleshooting section of the README

I forgot we had that! I think we can link to that from the error message. It should use a more stable link (commit hash or pinned to a version) than e.g. https://github.com/knurling-rs/probe-run#troubleshooting

Lotterleben, 2021-08-16:
I understand why we'd want a more stable link (e.g. if someone is using an old probe-run version, newer info might not help), but how do we ensure we remember to bump this link regularly?

Urhengulas, 2021-08-19:
> It should use a more stable link (commit hash or pinned to a version) than e.g. https://github.com/knurling-rs/probe-run#troubleshooting

I think linking to the headline of the troubleshooting section should be fine, since this won't change too often. If it would be a link to a sub-(sub-)headline I'd agree.

closed 2021-09-27 by bors[bot]