knurling-rs/probe-run #241: improve error message when user supplies an invalid chip name

Lotterleben, 2021-07-20:
**Describe the bug**
Currently, when I supply an incomplete or wrong chip name, for example

```console
probe-run --chip nrf5 tests/test_elfs/hello
```

, I get a somewhat misleading error message:

```console
➜  probe-run git:(main) ✗ probe-run --chip nrf5 tests/test_elfs/hello
(HOST) INFO  flashing program (5.75 KiB)
Error: Error while flashing

Caused by:
    The execution of 'erase_sector' failed with code 1. Perhaps your chip has write protected sectors that need to be cleared? Perhaps you need the --nmagic linker arg https://github.com/rust-embedded/cortex-m-quickstart/pull/95
```

I would expect to get a notification that what I passed to `--chip` isn't valid instead.

jonas-schievink, 2021-07-20:
I think what's happening here is that probe-rs picks the first target whose name starts with `nrf5`, and then that ends up being the wrong one (nRF51?), so flashing doesn't work.

I think what we should do here is adjust the code below to check if the fetched `Target`'s name matches what the user input, and issue an `info!` log that says that the target was adjusted if not (or some other on-by-default level):

https://github.com/knurling-rs/probe-run/blob/783724ad289f55883eb0ec43536ec2377b4c2a49/src/target_info.rs#L17

japaric, 2021-07-21:
the alternative is to require an exact name match but this would be a breaking change (we are releasing probe-run 0.3 soon-ish so that's not much of an issue). some full chip names are rather non-memorable though, e.g. instead `--chip nrf52840` you would need `--chip nRF52840_xxAA`.

OTOH, for the start substring `nrf52832` there are actually *two* matches in probe-rs database (see `probe-run --list-chips`): `nRF52832_xxAA` and `nRF52832_xxBB`. unsure what the difference between the 2 is but if you use `probe-run --chip nrf52832` today you *do not get an errror*, `probe-rs` probably uses the first match? (not sure). I think that would be worthwhile to change to emit an error if there's more than one match (which also is a breaking change AND it technically turns probe-rs additions to their target database that generate ambiguity into breaking changes in probe-run)

Urhengulas, 2022-04-05:
I discovered that `probe-rs` does not accept ambiguous chip names anymore since https://github.com/probe-rs/probe-rs/pull/815, so I think we can close this issue here.

closed 2022-04-05 by Urhengulas