knurling-rs/probe-run #243: new release for probe-rs v0.11?

newAM, 2021-07-31:
I was debugging an issue with the folks in the `probe-rs` matrix chat, and the resolution was just installing `probe-run` from the `main` branch on git because it contains `probe-rs` v0.11 instead of v0.10 in the current v0.2.4 `probe-run` release.

Would it be possible to get a new `probe-run` release with `probe-rs` v0.11?  It would help resolve some bugs in the current release for people installing from `crates.io`.

justahero, 2021-08-02:
Hello @newAM ,

we released `probe-run` version `0.2.5` today. The major change is the version bump of `probe-rs` to version `v0.11`.

Hope that resolves your issue.

newAM, 2021-08-02:
That fixes it, thank you!

closed 2021-08-02 by newAM