knurling-rs/probe-run #255: Timeout on every second run

folkertdev, 2021-09-07:
**Describe the bug**
Every second run of `probe-run` results in a timeout

```
    Finished dev [optimized + debuginfo] target(s) in 0.02s
     Running `probe-run --chip STM32WLE5JBIx target/thumbv7em-none-eabi/debug/pingpong --probe 001A00195553500720393256`
(HOST) INFO  flashing program (18.83 KiB)
(HOST) INFO  success!
Error: Operation timed out
```
When there is no timeout, the program runs just fine. It is consistently every second time that a timeout occurs.

**To Reproduce**
Steps to reproduce the behavior:

For me, flash the supplied elf file to the device (in my case, an stm32wle). Every second time, this operation will timeout. The other time it will run just fine.

**config.toml**
The contents of your project's `.cargo/config.toml` file

``` toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip STM32WLE5JBIx"

rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C", "link-arg=--nmagic",
]

[build]
target = "thumbv7em-none-eabi"   # Cortex-M4 and Cortex-M7 (no FPU)

[alias]
rb = "run --bin"
rrb = "run --release --bin"

```

**Probe details**
You can get this information from [`probe-rs-cli`](https://crates.io/crates/probe-rs-cli). Your microcontroller must be connected to your PC / laptop when you run the command below.

*Example:*

``` console
[0]: STLink V3 (VID: 0483, PID: 374e, Serial: 001A00195553500720393256, STLink)
```

**Operating System:**
Linux (mint 19)

**ELF file (attachment)**

[pingpong.tar.gz](https://github.com/knurling-rs/probe-run/files/7121583/pingpong.tar.gz)

**Additional context**

@jamesmunns  suggested that configurable timeouts/retries may be a solution here.

jonas-schievink, 2021-09-07:
Does this also happen with `cargo embed`?

folkertdev, 2021-09-07:
I'm not sure I understand. The flashing process always succeeds. Can `cargo embed` actually (attempt to) run the application? It's not clear how to do that from the `--help` message or the README

Urhengulas, 2021-09-15:
Hi @folkertdev, yes, `cargo-embed` also runs the application and can e.g. connect to rtt. `cargo-flash` only flashes the program.

Urhengulas, 2022-04-05:
Fixed by #285

closed 2022-04-05 by Urhengulas