knurling-rs/probe-run #278: probe-run counts BSS in size-to-be-flashed

Rahix, 2021-11-18:
**Describe the bug**
The size shown while flashing includes the `.bss` section of the binary.  I found this comment in PR #106:

> due to a bug in cortex-m-rt (which has already been fixed but the fix is not on crates.io) this will include .bss in the count. the bug also causes probe-rs to load a bunch of zeros, as many as the size of .bss, to Flash so in a sense the reported value is correct in that it reflects the amount of work probe-rs will do

I noticed that `cargo-embed` seems to properly account for sections in flash only so I would assume the same should be possible in `probe-run`?

**To Reproduce**
Flash any binary with a large `.bss` section and notice that it gets added into the shown size during flashing.

**Expected Behavior**
Only sections in flash are written and the size shown also reflects that.

jonathanpallant, 2021-11-23:
So #106 is curious, because we basically have to parse the ELF twice. probe-run does it to find the segment sizes (and also for its own purposes), and then probe-rs does it again to do the actual flashing. We need to either:

a) get the two to match (and ensure they remain in-sync), or
b) get probe-rs to do both.

probe-rs has a "dummy-run" mode, and it can report "progress". I can see two options:

1. We could do a first-pass in "dummy" mode and collate all the reports (number of pages programmed, number of blocks erased) and then inform the user how long the real thing will take.
2. We could do a single-pass and report real-time progress as the erasing/programming occurs.

jonathanpallant, 2021-11-23:
I propose a solution in https://github.com/knurling-rs/probe-run/pull/281.

jonathanpallant, 2021-11-24:
#281 was merged. Will be in next release.

closed 2021-11-24 by jonathanpallant