knurling-rs/probe-run #289: Regression: probe-run 0.3.1 fails to install on Ubuntu 20.04 LTS

D1plo1d, 2021-11-28:
**Describe the bug**

probe-run v0.3.1 fails to install on my system (Ubuntu 20.04 LTS) however v0.3.0 installs without issue.

**To Reproduce**
Steps to reproduce the behavior:

1. Using ubuntu 20.04 LTS install libusb-1.0.0-dev and libudev-dev:
```
libusb-1.0-0-dev is already the newest version (2:1.0.23-2build1).
libudev-dev is already the newest version (245.4-4ubuntu3.13).
```
2.  `cargo install probe-run --version 0.3.0`
3. This install succeeds without issue.
4.  `cargo install probe-run --version 0.3.1`
5. This install returns a ld error indicating none of the libusb functions can be found. This is a snippet, the actual error is quite long:
```bash
  = note: /usr/bin/ld: /tmp/cargo-installsMCyEP/release/deps/libprobe_rs-81ef77483e944b7b.rlib(probe_rs-81ef77483e944b7b.probe_rs.d6551148-cgu.10.rcgu.o): in function `core::ptr::drop_in_place<probe_rs::probe::stlink::StLink<probe_rs::probe::stlink::usb_interface::StLinkUsbDevice>>':
          probe_rs.d6551148-cgu.10:(.text._ZN4core3ptr115drop_in_place$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$GT$17hd8cb3aba2093ebeaE+0x1b): undefined reference to `libusb_release_interface'
          /usr/bin/ld: probe_rs.d6551148-cgu.10:(.text._ZN4core3ptr115drop_in_place$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$GT$17hd8cb3aba2093ebeaE+0x6b): undefined reference to `libusb_release_interface'
          /usr/bin/ld: probe_rs.d6551148-cgu.10:(.text._ZN4core3ptr115drop_in_place$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$GT$17hd8cb3aba2093ebeaE+0x99): undefined reference to `libusb_close'
          /usr/bin/ld: /tmp/cargo-installsMCyEP/release/deps/libprobe_rs-81ef77483e944b7b.rlib(probe_rs-81ef77483e944b7b.probe_rs.d6551148-cgu.10.rcgu.o): in function `core::ptr::drop_in_place<probe_rs::probe::jlink::JLink>':
          probe_rs.d6551148-cgu.10:(.text._ZN4core3ptr50drop_in_place$LT$probe_rs..probe..jlink..JLink$GT$17h11510fa310c582a4E+0x40): undefined reference to `libusb_release_interface'
```

D1plo1d, 2021-11-28:
This appears to be an `rusb` issue by way of `probe-rs`: https://github.com/a1ien/rusb/issues/117

The workaround for now is to remove older versions of libusb and then re-try installing probe-run:
`sudo apt remove libusb-dev libusb-0.1-4`

senden9, 2021-11-29:
Hi!
Thanks for the diagnosis. Had exactly the same problem on the same OS version. Your solution worked fine for me.

a1ien, 2022-04-27:
In libusb1-sys 0.6.2 this problem should be fixed

Urhengulas, 2022-07-25:
@a1ien [said](https://github.com/knurling-rs/probe-run/issues/289):
> In libusb1-sys 0.6.2 this problem should be fixed

Thank you! But I think we still need to wait for a new `rusb` version and then a new `probe-rs`.

Urhengulas, 2022-07-25:
> According to our lockfile, we're already using libusb1-sys 0.6.2.

That's nice. So I think we are done here :)

closed 2022-07-25 by Urhengulas

jonas-schievink, 2022-07-25:
Needs a probe-run release though

Urhengulas, 2022-07-25:
> Needs a probe-run release though

That's long overdue anyways 😉