knurling-rs/probe-run #292: Regression: install fails on Windows due to breakage in signal-hook 0.3.11

9names, 2021-12-08:
**Describe the bug**
cargo install probe-run fails to install on Windows 10

**To Reproduce**
cargo install probe-run

**Expected and observed behavior**
Expected behavior: installation completes successfully.
Observed behavior:
``` console
error[E0412]: cannot find type `sigset_t` in this scope
   --> C:\Users\9names\.cargo\registry\src\github.com-1ecc6299db9ec823\signal-hook-0.3.11\src\low_level\signal_details.rs:149:36
    |
149 | unsafe fn prepare_sigset(set: *mut sigset_t, signal: c_int) {
    |                                    ^^^^^^^^ not found in this scope

error[E0425]: cannot find function `sigemptyset` in crate `libc`
   --> C:\Users\9names\.cargo\registry\src\github.com-1ecc6299db9ec823\signal-hook-0.3.11\src\low_level\signal_details.rs:150:11
    |
150 |     libc::sigemptyset(set);
    |           ^^^^^^^^^^^ not found in `libc`

error[E0425]: cannot find function `sigaddset` in crate `libc`
   --> C:\Users\9names\.cargo\registry\src\github.com-1ecc6299db9ec823\signal-hook-0.3.11\src\low_level\signal_details.rs:151:11
    |
151 |     libc::sigaddset(set, signal);
    |           ^^^^^^^^^ not found in `libc`

Some errors have detailed explanations: E0412, E0425.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `signal-hook` due to 3 previous errors
```

**Operating System:**
Windows 10

**Additional context**

Installation succeeds from a local clone of probe-run with signal-hook = "=0.3.10" specified in Cargo.toml

jonas-schievink, 2021-12-08:
Huh, does `cargo install` ignore the `Cargo.lock`? Because that's still on 0.3.10.

jonas-schievink, 2021-12-08:
Upstream issue: https://github.com/vorner/signal-hook/issues/125

Not sure if it makes sense to do anything here, but the lockfile is checked in to specifically prevent issues like that, weird that it doesn't work.

9names, 2021-12-08:
Working fine again now.
> does cargo install ignore the Cargo.lock

That is indeed weird now that you mention it. I might look into it this weekend - hopefully not a cargo bug.

closed 2021-12-08 by 9names