knurling-rs/probe-run #298: [feature] Support defmt without flashing via a new --promise-hardware-unchanged

danielzfranklin, 2022-01-31:
I'm debugging a finicky peripheral so I often want to ask ("If I hard power cycle everything does it start working"), and it feels wasteful to reflash each time. My understanding is probe-run requires a reflash for defmt because you might see subtly wrong logs if the elf on your harddrive didn't match the one flashed.

It's easy to change the elf on your harddrive without noticing (i.e. an editor configured to check on save), but I'm guessing most people would know if they reflashed their hardware or plugged in different hardware.

I'm proposing every time `probe-run` flashes it copies the elf to `{target}/probe_run/last_flash/{binary_name}`.

If you run a project with defmt logs without flashing and provide `--promise-hardware-unchanged` probe-run would use the cached elf.

Urhengulas, 2022-02-01:
Hi @danielzfranklin, Thank you for opening the issue!

I see your point and I also would agree that it should be obvious if you are using the wrong elf, if you are experienced with `defmt`. My only concern is that people just getting started with `defmt` (and maybe embedded development overall) could have a hard time figuring the problem out. But then it is the question why they would decide to not re-flash the elf.

My intuition is to ease the restrictions, by allowing to use `--no-flash` even if using defmt, but emitting a warning that this might cause problems.

I will get a second opinion, but what do you think about it?

danielzfranklin, 2022-02-01:
That sounds great.

commit 2dde3114 (2022-02-22): Merge #303

303: Don't bail, but only warn if using `--no-flash` with defmt. r=Urhengulas a=Urhengulas

This PR makes `probe-run` not bail when a user uses `defmt` together with the `--no-flash` cli-option, but only emits a warning.

This allows for more felxibility during development, but should only be used when you know what you are doing.

Fixes #298.