knurling-rs/probe-run #306: JLink Protocol Selection

kiranshila, 2022-03-01:
I'm trying to get probe-run to work and can't for the life of me figure out what's going wrong. I have an ATSAMD21E17A which communicates well with other tools (i.e. probe.rs) but it refuses to flash with probe-run.

**To Reproduce**
If I try to burn my project with
```console
cargo flash --chip ATSAMD21E17A --speed 400 --protocol SWD
```
It works as expected. However with probe-run

```console
$ cargo run
warning: unused manifest key: metadata
    Finished dev [optimized + debuginfo] target(s) in 0.02s
     Running `probe-run --chip ATSAMD21E17A --speed 400 target/thumbv6m-none-eabi/debug/u_wave_synth`
(HOST) INFO  flashing program (19 pages / 19.00 KiB)
Error: Error while flashing

Caused by:
    0: The page write of the page at address 0x00000800 failed.
    1: An error with the usage of the probe occured
    2: Operation timed out
```

I see the same behavior if I remove the `--protocol` bit from `cargo flash`. I'm not sure if probe-run is falling back to JTAG or what.

**config.toml**
``` toml
[target.thumbv6m-none-eabi]
runner = "probe-run --chip ATSAMD21E17A --speed 400"
rustflags = [
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=--nmagic",
  "-C", "link-arg=-Tdefmt.x",
]

[build]
target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+

```

**Probe details**

``` console
$ probe-rs-cli list
The following devices were found:
[0]: J-Link (J-Link) (VID: 1366, PID: 0101, Serial: 000801023684, JLink)
```

**Operating System:**
Arch

Urhengulas, 2022-03-07:
Hi @kiranshila, thank you for raising the issue.

I assume the problem is that `probe-rs` defaults to using the `probe_rs::WireProtocol::Jtag`, while you want to use `Swd`. And `probe-run` currently doesn't have a flag to to configure the protocol.

Could you please install following branch where I am setting the protocol to SWD, to see if it works then?

```console
$ cargo install --git https://github.com/knurling-rs/probe-run --branch swd-protocol
```

If that fixes the problem we can easily add a new flag to configure that.

kiranshila, 2022-03-09:
Hmm, still no dice, I'm getting the same error

japaric, 2022-03-21:
I would suggest checking the logs of both programs and see where behavior diverges between probe-run and cargo-flash. with probe-run, pass `-vv` to it to see the logs from the `probe_rs` crate. for `cargo-flash`, I guess setting the `RUST_LOG=probe_rs=trace` env var should do the trick.

at least with SWD, there's quite a bit of communication with the probe before the 'flash' operation. if JTAG is similar, then I would expect a protocol mismatch (if that's indeed the issue) to fail much earlier than during the 'flash write', i.e. I would expect to see a different error message.

it could be that cargo-flash is doing some additional manipulation of flash write protection bits. at least with the nrf52840 that was necessary to unlock the flash before writing to it for the first time (chip was locked out of factory); these day that step is done in `probe-rs` so both `cargo-flash` and `probe-run` do the right thing with that chip.