knurling-rs/probe-run #307: Mac OS X: no probe was found

brandonros, 2022-03-02:
[image]

```
Brandons-MacBook-Air:rp2040-project-template brandonros 2022-03-02 17:50:45 $ probe-run --list-probes
Error: no probe was found.

Common reasons for this are faulty cables or missing permissions.
For detailed instructions, visit: https://github.com/knurling-rs/probe-run/tree/2f138c3#troubleshooting
```

```
$ elf2uf2-rs --verbose --deploy ./target/thumbv6m-none-eabi/debug/rp2040-project-template
Found pico uf2 disk /Volumes/RPI-RP2
Detected FLASH binary
Mapped segment 0x10000100->0x100001a8 (0x10000100->0x100001a8)
Mapped segment 0x10000000->0x10000100 (0x10000000->0x10000100)
Mapped segment 0x100001a8->0x10003208 (0x100001a8->0x10003208)
Mapped segment 0x10003210->0x10003a98 (0x10003210->0x10003a98)
Mapped segment 0x10003a98->0x10003ac8 (0x2003fbc8->0x2003fbf8)
Uninitialized segment 0x2003fbf8->0x20040000 (0x2003fbf8->0x20040000)
Transfering program to pico
Page 0 / 59 0x10000000
Page 1 / 59 0x10000100
Page 2 / 59 0x10000200
Page 3 / 59 0x10000300
Page 4 / 59 0x10000400
Page 5 / 59 0x10000500
Page 6 / 59 0x10000600
Page 7 / 59 0x10000700
Page 8 / 59 0x10000800
Page 9 / 59 0x10000900
Page 10 / 59 0x10000a00
Page 11 / 59 0x10000b00
Page 12 / 59 0x10000c00
Page 13 / 59 0x10000d00
Page 14 / 59 0x10000e00
Page 15 / 59 0x10000f00
Page 16 / 59 0x10001000
Page 17 / 59 0x10001100
Page 18 / 59 0x10001200
Page 19 / 59 0x10001300
Page 20 / 59 0x10001400
Page 21 / 59 0x10001500
Page 22 / 59 0x10001600
Page 23 / 59 0x10001700
Page 24 / 59 0x10001800
Page 25 / 59 0x10001900
Page 26 / 59 0x10001a00
Page 27 / 59 0x10001b00
Page 28 / 59 0x10001c00
Page 29 / 59 0x10001d00
Page 30 / 59 0x10001e00
Page 31 / 59 0x10001f00
Page 32 / 59 0x10002000
Page 33 / 59 0x10002100
Page 34 / 59 0x10002200
Page 35 / 59 0x10002300
Page 36 / 59 0x10002400
Page 37 / 59 0x10002500
Page 38 / 59 0x10002600
Page 39 / 59 0x10002700
Page 40 / 59 0x10002800
Page 41 / 59 0x10002900
Page 42 / 59 0x10002a00
Page 43 / 59 0x10002b00
Page 44 / 59 0x10002c00
Page 45 / 59 0x10002d00
Page 46 / 59 0x10002e00
Page 47 / 59 0x10002f00
Page 48 / 59 0x10003000
Page 49 / 59 0x10003100
Page 50 / 59 0x10003200
Page 51 / 59 0x10003300
Page 52 / 59 0x10003400
Page 53 / 59 0x10003500
Page 54 / 59 0x10003600
Page 55 / 59 0x10003700
Page 56 / 59 0x10003800
Page 57 / 59 0x10003900
Page 58 / 59 0x10003a00

```

elf2uf2-rs worked fine

jonathanpallant, 2022-03-14:
Hi! The RP2040 has a USB Mass Storage based ROM Bootloader. Unfortunately this is only suitable for uploading UF2 files, and cannot be used as a debugger with Probe Run. elf2uf2-rs works by finding where the drive is mounted and `cp`'ing the UF2 file to the mounted folder.

You will need a Raspberry Pico running DapperMime firmware, or some other supported Debug Probe to work with probe-run.

japaric, 2022-03-21:
PR #304 documents this more clearly.

commit 7e64a444 (2022-04-04): Merge #314

314: update README r=Urhengulas a=japaric

to answer questions we received recently

closes #309 
closes #307
commit 8e09e243 (2022-04-04): Merge #314

314: update README r=jonathanpallant a=japaric

to answer questions we received recently

closes #309 
closes #307
closed 2022-04-04 by bors[bot]