knurling-rs/probe-run #309: "stm32h723zg" board not available

imrank03, 2022-03-13:
Hi, I am working on a **`stm32h723zg`** board but this board is not available in the **prob-run chips list** so can you please guide me how to add this board in probe-run?

burrbull, 2022-03-13:
Update https://github.com/probe-rs/probe-rs/blob/master/probe-rs/targets/STM32F7_Series.yaml

burrbull, 2022-03-13:
Should be in probe-rs master already:
https://github.com/probe-rs/probe-rs/pull/1011

imrank03, 2022-03-13:
> Should be in probe-rs master already:
> [probe-rs/probe-rs#1011](https://github.com/probe-rs/probe-rs/pull/1011)

@burrbull Thank you for your response, but it is not showing in the chip list. Or I need to reinstall 'probe-run'.

imrank03, 2022-03-13:
@burrbull These are the variants available for the **stm32h7** series

`STM32H7 Series
    Variants:
        STM32H742AGIx
        STM32H742AIIx
        STM32H742BGTx
        STM32H742BITx
        STM32H742IGKx
        STM32H742IGTx
        STM32H742IIKx
        STM32H742IITx
        STM32H742VGHx
        STM32H742VGTx
        STM32H742VIHx
        STM32H742VITx
        STM32H742XGHx
        STM32H742XIHx
        STM32H742ZGTx
        STM32H742ZITx
        STM32H743AGIx
        STM32H743AIIx
        STM32H743BGTx
        STM32H743BITx
        STM32H743IGKx
        STM32H743IGTx
        STM32H743IIKx
        STM32H743IITx
        STM32H743VGHx
        STM32H743VGTx
        STM32H743VIHx
        STM32H743VITx
        STM32H743XGHx
        STM32H743XIHx
        STM32H743ZGTx
        STM32H743ZITx
        STM32H745BGTx
        STM32H745BITx
        STM32H745IGKx
        STM32H745IGTx
        STM32H745IIKx
        STM32H745IITx
        STM32H745XGHx
        STM32H745XIHx
        STM32H745ZGTx
        STM32H745ZITx
        STM32H747AGIx
        STM32H747AIIx
        STM32H747BGTx
        STM32H747BITx
        STM32H747IGTx
        STM32H747IITx
        STM32H747XGHx
        STM32H747XIHx
        STM32H747ZIYx
        STM32H750IBKx
        STM32H750IBTx
        STM32H750VBTx
        STM32H750XBHx
        STM32H750ZBTx
        STM32H753AIIx
        STM32H753BITx
        STM32H753IIKx
        STM32H753IITx
        STM32H753VIHx
        STM32H753VITx
        STM32H753XIHx
        STM32H753ZITx
        STM32H755BITx
        STM32H755IIKx
        STM32H755IITx
        STM32H755XIHx
        STM32H755ZITx
        STM32H757AIIx
        STM32H757BITx
        STM32H757IITx
        STM32H757XIHx
        STM32H757ZIYx
        STM32H7A3AGIxQ
        STM32H7A3AIIxQ
        STM32H7A3IGKx
        STM32H7A3IGKxQ
        STM32H7A3IGTx
        STM32H7A3IGTxQ
        STM32H7A3IIKx
        STM32H7A3IIKxQ
        STM32H7A3IITx
        STM32H7A3IITxQ
        STM32H7A3LGHxQ
        STM32H7A3LIHxQ
        STM32H7A3NGHx
        STM32H7A3NIHx
        STM32H7A3QIYxQ
        STM32H7A3RGTx
        STM32H7A3RITx
        STM32H7A3VGHx
        STM32H7A3VGHxQ
        STM32H7A3VGTx
        STM32H7A3VGTxQ
        STM32H7A3VIHx
        STM32H7A3VIHxQ
        STM32H7A3VITx
        STM32H7A3VITxQ
        STM32H7A3ZGTx
        STM32H7A3ZGTxQ
        STM32H7A3ZITx
        STM32H7A3ZITxQ
        STM32H7B0ABIxQ
        STM32H7B0IBKxQ
        STM32H7B0IBTx
        STM32H7B0RBTx
        STM32H7B0VBTx
        STM32H7B0ZBTx
        STM32H7B3AIIxQ
        STM32H7B3IIKx
        STM32H7B3IIKxQ
        STM32H7B3IITx
        STM32H7B3IITxQ
        STM32H7B3LIHxQ
        STM32H7B3NIHx
        STM32H7B3QIYxQ
        STM32H7B3RITx
        STM32H7B3VIHx
        STM32H7B3VIHxQ
        STM32H7B3VITx
        STM32H7B3VITxQ
        STM32H7B3ZITx
        STM32H7B3ZITxQ
`

burrbull, 2022-03-13:
You need to update `probe-run` to use `probe-rs` from git instead of released:
https://github.com/knurling-rs/probe-run/blob/2eb6eeb8010a0822bd25fcebb6bf36a97730fc3f/Cargo.toml#L27
And then install from source.
Or wait for new release.

burrbull, 2022-03-13:
Try to install from this branch:
```
cargo install probe-run --git https://github.com/tronical/probe-run --branch stm32h7_update --force
```

imrank03, 2022-03-13:
> Try to install from this branch:
> 
> ```
> cargo install probe-run --git https://github.com/tronical/probe-run --branch stm32h7_update --force
> ```

I am getting this error.

[image]

burrbull, 2022-03-13:
Ok. What about
```
cargo install probe-run --git https://github.com/burrbull/probe-run --branch h7 --force
```

imrank03, 2022-03-13:
@burrbull Thank you so much, now it's working.

imrank03, 2022-03-13:
My upcoming project, which I plan to use `atsaml21`, **probe-run** is not available to this board so how can I add?

burrbull, 2022-03-13:
All questions about chip support in this repo, please: https://github.com/probe-rs/probe-rs

Or https://matrix.to/#/#probe-rs:matrix.org

imrank03, 2022-03-13:
okay, thanks.

japaric, 2022-03-21:
Thanks for answering here @burrbull . PR #314 documents what to do when a chip is not supported by stock probe-run (i.e. how to build probe-run against the git version of probe-rs)

commit 7e64a444 (2022-04-04): Merge #314

314: update README r=Urhengulas a=japaric

to answer questions we received recently

closes #309 
closes #307
commit 8e09e243 (2022-04-04): Merge #314

314: update README r=jonathanpallant a=japaric

to answer questions we received recently

closes #309 
closes #307
closed 2022-04-04 by bors[bot]