knurling-rs/probe-run #325: Fails to build on Cygwin.

hydra, 2022-06-06:
**Describe the bug**
`cargo install probe-run` fails with the following error on Cygwin.

```
The following warnings were emitted during compilation:

warning: In file included from libusb\libusb/libusbi.h:92,
warning:                  from libusb\libusb/os/events_windows.c:23:
warning: libusb\libusb/os/threads_windows.h:63:8: error: redefinition of ‘struct timespec’
warning:    63 | struct timespec {
warning:       |        ^~~~~~~~
warning: In file included from /usr/include/sys/timespec.h:38,
warning:                  from /usr/include/sys/select.h:16,
warning:                  from /usr/include/sys/types.h:50,
warning:                  from libusb\libusb/libusb.h:40,
warning:                  from libusb\libusb/libusbi.h:38,
warning:                  from libusb\libusb/os/events_windows.c:23:
warning: /usr/include/sys/_timespec.h:47:8: note: originally defined here
warning:    47 | struct timespec {
warning:       |        ^~~~~~~~

error: failed to run custom build command for `libusb1-sys v0.5.0`

Caused by:
  process didn't exit successfully: `C:\cygwin64\tmp\cargo-install8xwhXG\release\build\libusb1-sys-2124665117aa9f86\build-script-build` (exit code: 1)
  --- stdout
  cargo:vendored=1
  cargo:static=1
  cargo:include=C:\cygwin64\tmp\cargo-install8xwhXG\release\build\libusb1-sys-d9eb4c839273dc98\out\include
  cargo:rustc-link-lib=dylib=user32
  TARGET = Some("x86_64-pc-windows-gnu")
  OPT_LEVEL = Some("3")
  HOST = Some("x86_64-pc-windows-gnu")
  CC_x86_64-pc-windows-gnu = None
  CC_x86_64_pc_windows_gnu = None
  HOST_CC = None
  CC = None
  CFLAGS_x86_64-pc-windows-gnu = None
  CFLAGS_x86_64_pc_windows_gnu = None
  HOST_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr,llvm14-builtins-abi,sse,sse2")
  running: "gcc.exe" "-O3" "-ffunction-sections" "-fdata-sections" "-m64" "-I" "C:\\cygwin64\\tmp\\cargo-install8xwhXG\\release\\build\\libusb1-sys-d9eb4c839273dc98\\out\\include" "-I" "libusb\\libusb" "-DPRINTF_FORMAT(a, b)=" "-DENABLE_LOGGING=1" "-DOS_WINDOWS=1" "-DDEFAULT_VISIBILITY=" "-DPLATFORM_WINDOWS=1" "-o" "C:\\cygwin64\\tmp\\cargo-install8xwhXG\\release\\build\\libusb1-sys-d9eb4c839273dc98\\out\\libusb\\libusb/os/events_windows.o" "-c" "libusb\\libusb/os/events_windows.c"
  cargo:warning=In file included from libusb\libusb/libusbi.h:92,
  cargo:warning=                 from libusb\libusb/os/events_windows.c:23:
  cargo:warning=libusb\libusb/os/threads_windows.h:63:8: error: redefinition of ‘struct timespec’
  cargo:warning=   63 | struct timespec {
  cargo:warning=      |        ^~~~~~~~
  cargo:warning=In file included from /usr/include/sys/timespec.h:38,
  cargo:warning=                 from /usr/include/sys/select.h:16,
  cargo:warning=                 from /usr/include/sys/types.h:50,
  cargo:warning=                 from libusb\libusb/libusb.h:40,
  cargo:warning=                 from libusb\libusb/libusbi.h:38,
  cargo:warning=                 from libusb\libusb/os/events_windows.c:23:
  cargo:warning=/usr/include/sys/_timespec.h:47:8: note: originally defined here
  cargo:warning=   47 | struct timespec {
  cargo:warning=      |        ^~~~~~~~
  exit code: 1

  --- stderr

  error occurred: Command "gcc.exe" "-O3" "-ffunction-sections" "-fdata-sections" "-m64" "-I" "C:\\cygwin64\\tmp\\cargo-install8xwhXG\\release\\build\\libusb1-sys-d9eb4c839273dc98\\out\\include" "-I" "libusb\\libusb" "-DPRINTF_FORMAT(a, b)=" "-DENABLE_LOGGING=1" "-DOS_WINDOWS=1" "-DDEFAULT_VISIBILITY=" "-DPLATFORM_WINDOWS=1" "-o" "C:\\cygwin64\\tmp\\cargo-install8xwhXG\\release\\build\\libusb1-sys-d9eb4c839273dc98\\out\\libusb\\libusb/os/events_windows.o" "-c" "libusb\\libusb/os/events_windows.c" with args "gcc.exe" did not execute successfully (status code exit code: 1).

warning: build failed, waiting for other jobs to finish...
error: failed to compile `probe-run v0.3.3`, intermediate artifacts can be found at `C:\cygwin64\tmp\cargo-install8xwhXG`
```

**To Reproduce**
[... 254 lines cut ...]
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Cannot open DFU device 1d50:6017 found on devnum 14 (LIBUSB_ERROR_NOT_SUPPORTED)
Found DFU: [0483:df11] ver=0200, devnum=14, cfg=1, intf=0, path="3-2.1.3", alt=0, name="@External Flash /0x90000000/998*128Kg,1*128Kg,4*128Kg,21*128Ka", serial="25001F000F51393035343038"
```

**Operating System:**
Windows 10 x64 + Cygwin x86_64.

**Additional context**
GCC:

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-cygwin/11/lto-wrapper.exe
Target: x86_64-pc-cygwin
Configured with: /mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-11.2.0/configure --srcdir=/mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-11.2.0 --prefix=/usr --exec-prefix=/usr --localstatedir=/var --sysconfdir=/etc --docdir=/usr/share/doc/gcc --htmldir=/usr/share/doc/gcc/html -C --build=x86_64-pc-cygwin --host=x86_64-pc-cygwin --target=x86_64-pc-cygwin --without-libiconv-prefix --without-libintl-prefix --libexecdir=/usr/lib --with-gcc-major-version-only --enable-shared --enable-shared-libgcc --enable-static --enable-version-specific-runtime-libs --enable-bootstrap --enable-__cxa_atexit --with-dwarf2 --with-tune=generic --disable-bootstrap --enable-languages=c,c++,fortran,lto,objc,obj-c++,jit --enable-graphite --enable-threads=posix --enable-libatomic --enable-libgomp --enable-libquadmath --enable-libquadmath-support --disable-libssp --enable-libada --disable-symvers --with-gnu-ld --with-gnu-as --with-cloog-include=/usr/include/cloog-isl --without-libiconv-prefix --without-libintl-prefix --with-system-zlib --enable-linker-build-id --with-default-libstdcxx-abi=gcc4-compatible --enable-libstdcxx-filesystem-ts
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (GCC)
```

hydra, 2022-06-06:
I noticed that some of my cygwin packages were out of date, I upgraded them all and retried with the same result

I also tried building the latest version of libusb1-sys after upgrading my packages but was greeted with a different error, reported here: https://github.com/a1ien/rusb/issues/136

Cygwin packages:
```
$ cygcheck -c
Cygwin Package Information
Package                  Version                         Status
_autorebase              001091-1                        OK
alternatives             1.3.30c-10                      OK
autoconf                 15-1                            OK
autoconf2.1              2.13-12                         OK
autoconf2.5              2.69-5                          OK
autoconf2.7              2.71-2                          OK
automake                 11-1                            OK
automake1.10             1.10.3-3                        OK
automake1.11             1.11.6-3                        OK
automake1.12             1.12.6-3                        OK
automake1.13             1.13.4-2                        OK
automake1.14             1.14.1-3                        OK
automake1.15             1.15.1-2                        OK
automake1.16             1.16.5-1                        OK
automake1.9              1.9.6-11                        OK
base-cygwin              3.8-1                           OK
base-files               4.3-3                           OK
bash                     4.4.12-3                        OK
bash-completion          2.7-1                           OK
binutils                 2.38-1                          OK
bzip2                    1.0.8-1                         OK
ca-certificates          2021.2.52-1                     OK
clang                    8.0.1-1                         OK
compiler-rt              8.0.1-1                         OK
coreutils                8.32-1                          OK
crypto-policies          20190218-1                      OK
csih                     0.9.11-1                        OK
curl                     7.83.1-1                        OK
cygrunsrv                1.62-1                          OK
cygutils                 1.4.17-1                        OK
cygwin                   3.3.5-1                         OK
cygwin-devel             3.3.5-1                         OK
dash                     0.5.11.5-1                      OK
diffutils                3.8-1                           OK
editrights               1.03-1                          OK
file                     5.41-2                          OK
findutils                4.9.0-1                         OK
gawk                     5.1.1-1                         OK
gcc-core                 11.3.0-1                        OK
gcc-g++                  11.3.0-1                        OK
getent                   2.18.90-4                       OK
git                      2.36.1-1                        OK
grep                     3.7-2                           OK
groff                    1.22.4-1                        OK
gzip                     1.12-1                          OK
hostname                 3.13-1                          OK
info                     6.8-2                           OK
ipc-utils                1.0-2                           OK
less                     590-1                           OK
libargp                  20110921-3                      OK
libatomic1               11.3.0-1                        OK
libattr1                 2.4.48-2                        OK
libblkid1                2.33.1-2                        OK
libbrotlicommon1         1.0.9-2                         OK
libbrotlidec1            1.0.9-2                         OK
libbz2_1                 1.0.8-1                         OK
libc++-devel             8.0.1-1                         OK
libc++1                  8.0.1-1                         OK
libc++abi-devel          8.0.1-1                         OK
libc++abi1               8.0.1-1                         OK
libcbor                  0.9.0-3                         OK
[... 123 lines cut ...]
run                      1.3.4-2                         OK
sed                      4.8-1                           OK
tar                      1.34-1                          OK
terminfo                 6.1-1.20190727                  OK
terminfo-extra           6.1-1.20190727                  OK
texinfo                  6.8-2                           OK
tree                     1.7.0-1                         OK
tzcode                   2022a-1                         OK
tzdata                   2022a-1                         OK
unzip                    6.0-17                          OK
util-linux               2.33.1-2                        OK
vim                      8.2.4372-1                      OK
vim-common               8.2.4372-1                      OK
vim-doc                  8.2.4372-1                      OK
vim-minimal              8.2.4372-1                      OK
w32api-headers           10.0.0-1                        OK
w32api-runtime           10.0.0-1                        OK
which                    2.20-2                          OK
windows-default-manifest 6.4-1                           OK
xxd                      8.2.4372-1                      OK
xz                       5.2.5-1                         OK
zip                      3.0-12                          OK
zlib0                    1.2.12-1                        OK
zstd                     1.5.2-1                         OK
```

japaric, 2022-06-08:
Thanks for the report. Sadly, this is a problem in a dependency that we can't address. libusb1-sys is an indirect dependency and we have no means to bump its version; and as you have mentioned the git version of libusb1-sys does not compile on Cygwin either :-(

hydra, 2022-06-18:
Perhaps anyone reading this could add a +1 command or vote to the issue I raised (https://github.com/a1ien/rusb/issues/136) so that a little extra pressure is applied to the maintainers of the dependency?