knurling-rs/probe-run #332: v0.3.3 from crates.io does not install on current Debian Linux

BartMassey, 2022-07-25:
Typing `cargo install probe-run` on current Debian Linux x86_64 fails during the link stage, apparently because it cannot find `libusb`; `libusb-1.0-0-dev` is correctly installed.

Installing from top-of-tree on the `probe-run` repo (commit bcaf997) via `cargo install --path .` works fine. Installing from the `v0.3.3` tag (commit 9226307) fails the same way as with installing from `crates.io`.

I have not bisected this, although I can if you have trouble reproducing. A failed build run is attached:
[`build-errors.txt`](https://github.com/knurling-rs/probe-run/files/9177114/build-errors.txt)

Urhengulas, 2022-07-25:
Thank you for the report. This is a known issue with `rusb`. Please refer to https://github.com/knurling-rs/probe-run/issues/289 for details.

closed 2022-07-25 by Urhengulas

BartMassey, 2022-07-25:
Apologies: looked at that issue before reporting, but didn't catch that it was the same. Thanks!

Urhengulas, 2022-07-26:
No worries. It's the same root problem, but with different versions. As written in the other issue we still need a new release to fix it.