knurling-rs/probe-run #338: `actions-rs` is unmaintained

jonas-schievink, 2022-07-29:
CI currently uses the `actions-rs/toolchain` action to install Rust, but the project is unmaintained: https://github.com/actions-rs/toolchain/issues/216

GHA comes with `rustup` preinstalled though, so we can just invoke that to install Rust instead.

BriocheBerlin, 2022-10-18:
This seems to be already done. Should we close?

jonas-schievink, 2022-10-18:
It's still used:

https://github.com/knurling-rs/probe-run/blob/8add0cf89438cb619e7ef7a26256805202eabe7e/.github/workflows/ci.yml#L46

Urhengulas, 2022-11-04:
@BriocheBerlin said:
> This seems to be already done. Should we close?

So far I only updated it for `flip-link`, but there is nothing stopping us from implementing it in `probe-rs` and `defmt` as well.

closed 2023-02-08 by Urhengulas