knurling-rs/probe-run #348: Probe Not Found - Did all the roubleshooting steps

iot-resister, 2022-08-20:
I have an nRF52840. Its the first time using it so it could be a bad board.  I'm trying to follow the [testing hals](https://ferrous-systems.com/blog/defmt-test-hal/) blog.   I used the same cable for `cargo espflash` commands and it works fine.  Here are my permissions

```shell
lsusb | grep SEGGER
Bus 001 Device 104: ID 1366:1051 SEGGER J-L
ls -l /dev/bus/usb/001/104
crw-rw----+ 1 root root 189, 103 Aug 20 16:00 /dev/bus/usb/001/104
```

```
cat /etc/udev/rules.d/50-oxidize-global.rules
# nRF52840 Dongle in bootloader mode
ATTRS{idVendor}=="1915", ATTRS{idProduct}=="521f", TAG+="uaccess"

# nRF52840 Dongle applications
ATTRS{idVendor}=="2020", TAG+="uaccess"

# nRF52840 Development Kit
ATTRS{idVendor}=="1366", ENV{ID_MM_DEVICE_IGNORE}="1", TAG+="uaccess"
```

I also tried chown cause I needed to do that for esp32.  I'm on arch linux btw.  
```shell
sudo chown $USER /dev/bus/usb/001/104
```

Here's is my code:
https://github.com/iot-resister/probe-debug

I'm simply trying to run `cargo test` from the `self-tests` directory. 

Thanks for the awesome work!

iot-resister, 2022-08-20:
I forgot to add the out of  list probes:

```shell
probe-run --list-probes
the following probes were found:
[0]: J-Link (J-Link) (VID: 1366, PID: 1051, Serial: 001050201176, JLink)
```

iot-resister, 2022-08-23:
I'm getting a replacement from mouser. See if that does the trick.

Urhengulas, 2022-08-26:
Hi @iot-resister,

What error message do you get?

iot-resister, 2022-08-28:
@Urhengulas  thanks for responding.

```
➜  self-tests git:(main) cargo test --verbose
     Running `probe-run --chip nRF52840_xxAA --probe '1366:1015' /home/resister/play-nrf/target/thumbv7em-none-eabihf/debug/deps/gpio_input_floating-4e58c2baac985cf1`
Error: no probe was found.

Common reasons for this are faulty cables or missing permissions.
For detailed instructions, visit: https://github.com/knurling-rs/probe-run/tree/2f138c3#troubleshooting
error: test failed, to rerun pass '--test gpio-input-floating'

```

Urhengulas, 2022-08-29:
Let's see if you new probe does the trick.

joe-narvaez, 2022-08-30:
Same error with a new board. Maybe there are some steps I need to take to initialize the board?  Pre-reqs so probe-run can work? I'm brand new to rust and this board. I may have missed an important step, like adding a jumper or toggling a physical switch?

iot-resister, 2022-08-30:
Same issue with brand new board. I'm new to rust and this board. Maybe there are some steps I need to take to initialize the board? Pre-reqs to make it ready for probe-run?   Maybe a jumper or toggle a physical switch?   Maybe install a dependency?  I wish I could get more info or logs of what happens when I run `cargo test`

iot-resister, 2022-08-30:
I went through the following guide: 
https://session20q4.ferrous-systems.com/defmt_setup.html

Same error :(

closed 2022-08-30 by iot-resister

iot-resister, 2022-08-30:
Unlocking it via `nrf-recover` did the trick. if you use the the on/off switch it will reset the lock. simply unplugging will not.

Urhengulas, 2022-09-01:
I do have the same board, but never hat this problem 🤔  Glad that you could solve it!