knurling-rs/probe-run #354: Support for viewing defmt RTT logs without resetting device

cstrahan, 2022-09-25:
Hello,

Is there a means (or would be possible to add support) for a way to view `defmt` logs while a micro controller is already running (i.e. without resetting it first)? My current use case looks something like:

* I have a Raspberry Pi Pico
* I can bring the RUN pin low to reset the device
* When I do, sometimes the program resets and starts running from the beginning successfully...
* ... and other times, the device appears to lock up or crash -- the LCD I have attached as part of my project freezes, but I don't know if that's because the LCD controller is stuck in a bad state while everything else on the RP2040 is chugging along, or if something `panic`ed, or...

Logs here would be a big help.

I can do `probe-run target/thumbv6m-none-eabi/debug/pim --no-flash --chip rp2040`, which doesn't flash the device, but it _does_ reset it. It would help in this sort of scenario if I could view the `defmt` logs _without_ resetting the device.

I guess such a feature contradicts the "run" in the name `probe-run`, but unless there's already another executable that provides this ability, it would be really nice to have. Or if this is a fundamental limitation of `defmt`/`probe-run`, it would be nice to have an advisory in the docs (assuming I haven't overlooked such).

Thanks!

cstrahan, 2022-09-27:
Here's where I've described how to use `defmt` with `cargo-embed`: https://github.com/probe-rs/probe-rs/discussions/1237#discussioncomment-3727155

Still would be cool for `probe-run` to support this, as (at the very least) I really like the unique features `probe-run` has like colorization for the various log components (making it easier to spot log level, file names, etc) and support for full back traces.

whitequark, 2023-06-21:
https://github.com/knurling-rs/probe-run/pull/414