knurling-rs/probe-run #355: probe-run missing permissions on m1 Mac.

letto4135, 2022-09-25:
After following https://github.com/rp-rs/rp2040-project-template
the project fails to run, on 2 different pico models and a waveshare rp2040-plus and several different cables. The RP2040 shows up in the finder, so it is a data cable.
No documentation or issues that I can find on mac, only linux and windows.

```
probe-run --list-probes
Error: no probe was found.

Common reasons for this are faulty cables or missing permissions.
For detailed instructions, visit: https://github.com/knurling-rs/probe-run/tree/2f138c3#troubleshooting
```

Running `probe-run --chip RP2040 --list-probes` also results in the same issue.

Note: the instructions on the linked repo  for elf2uf2-rs do work, so not sure what the issue is with probe-run..

jonathanpallant, 2022-09-26:
Which debug probe are you using?

Quoting from the rp2040-project-template README:

```quote
If you aren't using a debugger, check out [alternative runners](https://github.com/knurling-rs/probe-run/issues/355#alternative-runners) for other options
```

Note that there is a difference between loading the flash through the RP2040 ROM bootloader in USB Mass Storage Device mode (which is what elf2uf2-rs uses), and using a debugger to program Flash and control the CPU core via the SWD/SWCLK pins (which probe-run requires).

It is possible to get the RP2040 to appear as its own USB debugger - the debugger code runs on one core and you get to program the other. I haven't tried that approach though and there are plenty of reasons why it might not work so I would always recommend a hardware debugger.

letto4135, 2022-10-03:
Yup my bad, didn't see the debugger part.

closed 2022-10-03 by letto4135